T-vK/ESP32-BLE-Keyboard #99: BatteryLevel not sent

rtausend, 2021-07-15:
Hi there!

i implemented blekeyboard on a TinyPICO and try to send the Batterylevel via 

`bleKeyboard.setBatteryLevel(batterypercent);`

But it only rarley get sent to my Windows11 Workstation ... But sometimes it seems to get transmitted ...

Any idea whats wrong there?

Thanks a lot, Robert

T-vK, 2021-07-15:
I don't know, it has never been tested with Windows 11 before.

rtausend, 2021-07-15:
Hi, same Problem on Windows 10

Holen Sie sich Outlook für Android<https://aka.ms/AAb9ysg>

galister, 2021-07-22:
For me it seems to only update the battery percentage upon pairing. I don't think I've seen it ever update otherwise, even after power cycling the BT device.

This is on Win10

rtausend, 2021-07-29:
Hmmm sometimes it updates it, but really not stable ... Any idea what i could try here?

T-vK, 2021-07-29:
I don't know to be honest. Maybe the value is only transmitted when certain events happen. 
I haven't actually tested that feature at all beyond setting the initial battery level. 
Is this only happening on Windows or does it also not update correctly on Android, iOS, Mac OS and Linux?

rtausend, 2021-07-30:
Hey! I only tried it on Windows till now, on MacOS i don't even see the device in the BT Menu ... will open another Case for that and see if there is any solution... Thanks

beastdjw, 2021-08-30:
I use this to read the latest battery charged informations status:

BG2CRW, 2022-05-06:
oh, I fix this bug, in BLEHIDDevice.cpp #45, you need to change this

this is origon:

	BLE2904* batteryLevelDescriptor = new BLE2904();
	batteryLevelDescriptor->setFormat(BLE2904::FORMAT_UINT8);
	batteryLevelDescriptor->setNamespace(1);
	batteryLevelDescriptor->setUnit(0x27ad);

	m_batteryLevelCharacteristic = m_batteryService->createCharacteristic((uint16_t) 0x2a19, BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_NOTIFY);
	m_batteryLevelCharacteristic->addDescriptor(batteryLevelDescriptor);
	m_batteryLevelCharacteristic->addDescriptor(new BLE2902());

this is my fix:

	BLE2904* batteryLevelDescriptor = new BLE2904();
	batteryLevelDescriptor->setFormat(BLE2904::FORMAT_UINT8);
	batteryLevelDescriptor->setNamespace(1);
	batteryLevelDescriptor->setUnit(0x27ad);

	m_batteryLevelCharacteristic = m_batteryService->createCharacteristic((uint16_t) 0x2a19, BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_NOTIFY);
	m_batteryLevelCharacteristic->addDescriptor(batteryLevelDescriptor);
	m_batteryLevelCharacteristic->addDescriptor(new BLE2904());

SuGlider, 2022-06-14:
I've submitted a fix in the ESP Arduino GH
https://github.com/espressif/arduino-esp32/pull/6864 

This shall close this issue as well, I guess.

skotl, 2022-08-17:
Would this fix be only in the base (original) BLE library? I'm having exactly the same issue using nimBLE - battery report works only in the constructor and not via `bleKeyboard.setBatteryLevel()`

ignis32, 2024-05-12:
@skotl Thanks a lot, that helped.