T-vK/ESP32-BLE-Keyboard #106: This support PS4 and PS5?

uecken, 2021-08-18:
This support PS4 and PS5?
I have only PS3 and this ESP32 controller cannot connect to PS3 due to PS3 cannot detects ESP32.
Anyone try to connect PS4 and PS5 ?
If so I'm glad anyone tell me.

T-vK, 2021-08-18:
Personally I haven't tried it because I don't own any Playstation

eingespieltx, 2022-07-15:
I tried it but its not working for me. Note that Im using it with ESPHome.

xelprep, 2024-04-22:
I've tried the latest release of this library with PS5 and was unable to make it work. I made some tweaks to test different auth methods with both the default and NimBLE backend. Seems like the PS5 doesn't agree with Secure Connect being enabled. I also tried with "Just Works" auth as well as "Display Only", "Display + Yes/No", and "Keyboard" IO capabilities defined. Using NimBLE debug logging it appears that the PS5 is properly authenticating the connection but it immediate drops the connection afterwards and reports the device isn't supported. I've even cloned an Apple Magic Keyboard attributes (name, VID, PID) and same behavior. There's got to be some attribute that the PS5 deems unworthy of connection.

osnwt, 2024-05-23:
Same here: PS5 does not establish the connection to such device.