T-vK/ESP32-BLE-Keyboard #113: Trying to emulate iOS bluetooth keyboard Siri Trigger

DevelopIdeas, 2021-09-12:
I am trying to replicate the "lock" button on my iOS bluetooth keyboard, reason being, when held down for 2 seconds it triggers Siri on the iPhone and iPad.
I've used pynput to determine what key is being sent (as it also works on the mac to bring the shutdown menu up).
This is the Keycode that the script logs:
<127> released

After much testing different methods I'm unable to get this library to send the same key code.

Any help would be appreciated.
I've also included the key listener script for convenience.

sploders101, 2022-06-21:
I know this is kind of an old post, but I've been investigating the same thing for the past few days, and I finally figured it out. Most bluetooth remotes bind siri to holding the play/pause button, so I tried just sending an extended play/pause press and it worked! For reference, I have an iPhone 13, which doesn't have a home button.

itspatch, 2023-04-10:
@sploders101 I was trying this using code that looks like this:

`bleKeyboard.press(KEY_MEDIA_PLAY_PAUSE);
  delay(2000);
  bleKeyboard.releaseAll(); `

However, this just play paused the media. Is this what you did to do an "extended play/pause press"?

sploders101, 2023-04-10:
@itspatch Unfortunately, it appears I've lost the source code for the project I was working on. I should have a backup from my old laptop, but it's not readily accessible. I'll see if I can track it down tonight or tomorrow afternoon.

That is essentially what I did though. I also made some modifications to the HID descriptor that may have helped. I'll post those when I find them, too.

itspatch, 2023-04-11:
@sploders101 thank you! I've been trying a lot of different combination of things but no dice. The modifications to the HID descriptor is probably the missing link for me. Thanks for looking into this!

sploders101, 2023-04-17:
@itspatch Sorry for taking so long. It's been a busy week. I finally found my old sketch. Here is a snippet of the HID descriptor I used. It also appears there was indeed a separate Siri key. Now that I think about it, I ended up doing a significant amount of research on this. HID pages are pretty difficult to figure out. It seems like there aren't really any guides to how they work; everyone just copies something they found from another tutorial without actually understanding it. Now that I know what I'm looking for, I found this pretty quickly. Section 15 has the consumer page documentation with a voice command button.

https://www.usb.org/sites/default/files/hut1_4.pdf

```cpp
// Report IDs:
#define KEYBOARD_ID 0x01
#define MEDIA_KEYS_ID 0x02

static const uint8_t _hidReportDescriptor[] = {
  USAGE_PAGE(1),      0x0C,          // USAGE_PAGE (Consumer)
  USAGE(1),           0x01,          // USAGE (Consumer Control)
  COLLECTION(1),      0x01,          // COLLECTION (Application)
  REPORT_ID(1),       MEDIA_KEYS_ID, //   REPORT_ID (3)
  USAGE_PAGE(1),      0x0C,          //   USAGE_PAGE (Consumer)
  LOGICAL_MINIMUM(1), 0x00,          //   LOGICAL_MINIMUM (0)
  LOGICAL_MAXIMUM(1), 0x01,          //   LOGICAL_MAXIMUM (1)
  REPORT_SIZE(1),     0x01,          //   REPORT_SIZE (1)
  REPORT_COUNT(1),    0x08,          //   REPORT_COUNT (8)
  USAGE(1),           0xB5,          //   USAGE (Scan Next Track)     ; bit 0: 1
  USAGE(1),           0xB6,          //   USAGE (Scan Previous Track) ; bit 1: 2
  USAGE(1),           0xB7,          //   USAGE (Stop)                ; bit 2: 4
  USAGE(1),           0xCD,          //   USAGE (Play/Pause)          ; bit 3: 8
  USAGE(1),           0xE2,          //   USAGE (Mute)                ; bit 4: 16
  USAGE(1),           0xE9,          //   USAGE (Volume Increment)    ; bit 5: 32
  USAGE(1),           0xEA,          //   USAGE (Volume Decrement)    ; bit 6: 64
  USAGE(1),           0xCF,          //   USAGE (Siri)                ; bit 7: 128
  HIDINPUT(1),        0x02,          //   INPUT (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
  END_COLLECTION(0)                  // END_COLLECTION
};
```

and here is the Siri button

```cpp
// Define Siri Key
const MediaKeyReport KEY_SIRI = {128, 0};

bleKeyboard.press(KEY_SIRI);
// vTaskDelay(pdMS_TO_TICKS( 500 )); // FreeRTOS task-based sleep
delay(500); // More common method of sleeping
bleKeyboard.release(KEY_SIRI);
```

If you have any questions, feel free to let me know. I'm about to try and rewrite all this in Rust so I can use a language I'm a little more comfortable with on this project, so it'll be good for me to refresh my memory on it anyway.

itspatch, 2023-04-18:
@sploders101 thanks so much, this worked! Really appreciate you taking the time to dig that out.

s00500, 2025-02-20:
hm... i am on ios 18.1.1 but this seem to not work for me... just does not do anything... any hints or ideas ?