T-vK/ESP32-BLE-Keyboard #114: print Extended ASCII Codes

tahunus, 2021-09-12:
Is there a way to bleKeyboard.print a character beyond ASCII (decimal) 127? I'm looking at foreign language characters (ASCII 128 thru 165) and specifically at character "Ñ" (ASCII 165).  

I'm building a wireless RPM Letterboard (a tool for autistic people) that's basically a wireless keyboard. The letter "Ñ" is part of the Spanish language (and on Spanish keyboards). I've got everything running except for this one letter.

Thanks for any insights!

(not an issue, help wanted)

T-vK, 2021-09-12:
Check the Internet for how its done with the yuDB Keyboard library. It should be the same for this library.

tahunus, 2021-10-01:
> Check the Internet for how its done with the USB Keyboard library. It should be the same for this library.

Finally found some info: keyboard.cpp (and your BLEkeyboard.cpp) specifically filter anything above ASCI 128 as a keyboard modifier, so mapping extended ASCII characters (i.e. international characters) would entail a massive modification of keyboard.cpp.  Since the mapping in said file is for the position of the keypress on a Std US keyboard, a workaround (albeit not a pretty one) is to issue the key that's physically on the (in this case) Spanish keyboard and have the receiving agent (in this case a notepad app) configured in said language. So, by issuing a ";" key, a "ñ" is printed since both sit in the same physical position (the first in the Std US keyb and the second in the Std Spanish keyb). Tks!

closed 2021-10-01 by tahunus

sivar2311, 2021-10-01:
Hi @tahunus 
Today I started looking for ways to change the keyboard layout.
I am currently working on a way to accept ascii codes > 127 and change the layout without having to modify the BleKeyboard.cpp.

Status now:
The layout can be changed at any time (even during runtime).
This is a work in progress and is not final and not (yet) approved by T-vK.
The implementation and function names might still change.
You might want to [take a look](https://github.com/sivar2311/ESP32-BLE-Keyboard/blob/Layout/examples/GermanKeyboard/GermanKeyboard.ino)

tahunus, 2022-08-15:
Hi @sivar2311 
Did you drop your project for extended ASCII codes?  I can't seen to find it in your repositories.

sivar2311, 2022-08-15:
Hello @tahunus 
Yes, I have tidied up my repositories a bit and deleted old stuff that I am not actively working on at the moment.