T-vK/ESP32-BLE-Keyboard #115: Keyboard stops working after reconnect

MikeX7, 2021-09-16:
Hi, I am using the  0.3.0 version with Windows 10 and I noticed, that when I add after I add the esp32 keyboard as a new bluetooth device on my windows computer, everything works just fine, the keys work and all, but if the bluetooth conenction is lost, because i turn the esp32 off and on for example, then even though the windows reconnects to the esp32 and it changes status from PAIRED to CONNECTED under the bluetooth devices, the keyboard no longer works and the only way to make it work again is to remove the esp32 keyboard from bluetooth devices on windows and then add it again, after which it works until the next time signal is lost.

I tried downgrading to 0.2.3 which works flawlessly and doesn't have this bug.

sivar2311, 2021-09-16:
Hi @MikeX7 

I have not noticed that. Thanks for pointing it out.
It looks like it is coming from my revision.
The problem is identified and I already have a solution for it.

As soon as I finalize this I will open a pull request.

sivar2311, 2021-09-17:
@MikeX7 please try [sivar2311/ESP32-BLE-Keyboard](https://github.com/sivar2311/ESP32-BLE-Keyboard)

This should work in both modes (classic and NimBLE)

sivar2311, 2021-09-20:
@MikeX7 did you had the chance to test the fix?

Neuroquila-n8fall, 2021-09-20:
@sivar2311  I ran into the exact same problem. Your fix seems to solve it though I had to pair the devices again once more.

sivar2311, 2021-09-20:
@Neuroquila-n8fall Thanks for the feedback. 
Did you test both "modes" (classic / NimBle)?

The solution was to change the bonding parameters.
Since the bonding parameters have changed, a new pairing is required. 
However, this should only be necessary once.

p.s.: Schöne Grüße aus Schleswig-Holstein :)

Neuroquila-n8fall, 2021-09-20:
I only checked the NimBle stuff since the other is just too much of a space hog...why bother...
Well, sounds reasonable enough :)

...grüße zurück aus dem Saarland :D

closed 2021-10-01 by T-vK

jdeltoft, 2022-09-10:
I tried this 0.3.1 beta, for the issue I just entered in #189. Unfortunately I wasn't able to get media keys working at all.  Same as I originally found with 0.3.2.  The exact same script though works fine for media keys with 0.3.0 except the one issue I entered in #189. Any ideas?

jnsw, 2023-01-15:
I just wrote code to press a Button, but after power cycling the device it says it connected again in windows 10, but won’t register any keystrokes anymore, I have to delete the device and reconnect it

LukaszChrzanowski, 2023-02-04:
Same here, but with Windows 11. Therefore I don't know if this will be some clue but I also use Lolin32 (but lite).

ekoslav, 2023-02-04:
I had same "disconnect" problem on Win 11, however, once I enabled the NimBLE mode, the connection is stable and if any of devices reboot, it automatically connects and works. The only (small) downside is that the battery level cannot be changed once set.

LukaszChrzanowski, 2023-02-04:
Tried that added `#define USE_NIBLE` before `#include <BleKeyboard.h>` but with no luck :/ My device is working only first time on pairing :/

GaetanoDiRusso, 2024-01-03:
Where did you get the code to fix the issue? I'm facing the same problem and I was not able to solve it

Louis467, 2024-04-07:
> Tried that added `#define USE_NIBLE` before `#include <BleKeyboard.h>` but with no luck :/ My device is working only first time on pairing :/
> 
> [EDIT] Ok - I took code from beta and now works like a charm. Thank You! :)

How did you get the code? I need help

S-procoder, 2025-06-30:
I am unable to pair my Doit Esp32 Devkit V1 using bleKeyboard.h library to my Laptop Windows 11 HP intel CORE i3 I am a beginner in bluetooth technology because I mostly use Wifi and Serial communication

PR #117 (2021-10-01): Fix for #115
This fixes #115