T-vK/ESP32-BLE-Keyboard #116: Reconnect after flashing

cvanelteren, 2021-09-20:
Hi, 

Thanks for the work on the BLE, it's helped a lot for my project. I have been running into an issue that after reflashing the esp32, I have to remove the device from bluetooth devices. Do you have any clues on what could be the cause for this?

# Description of problem
- Connect with esp32 on bluetooth
- [change some code & flash device]
- device constantly connects and reconnects

# Workaround
- Remove / forget device
- Reconnect / pair

T-vK, 2021-09-20:
I'm not sure, but maybe Linux gets confused if you change the device name because it still remembers the old name. Could that be the issue?

cvanelteren, 2021-09-20:
The device name / manufacturer name stays the same. I assume the bluetooth connects on mac address which would make it unlikely that it it gets confused. Bluetooth has been a bit of a journey on linux.  It is very peculiar! Anyhoozle this does not occur on your os?

JpMaster0, 2021-09-21:
Got the same behaviour on Win10 - guess thats normal for bluetooth

sivar2311, 2021-09-21:
As far as I know, "a few more" parameters are used for pairing (cryptographic keys, for example). My guess is that these are deleted / changed by flashing the device and thus a new pairing is required.

If the ESP is reflashed it is in my opinion a "new device" and I would even assume such behavior. 

Therefore, I can not consider this as an issue.

cvanelteren, 2021-09-21:
That makes sense! I will curb my flashing behaviors ;-). Thanks all!

closed 2021-09-21 by cvanelteren

cvanelteren, 2021-09-26:
I switched to NimBLE using the recent support and now the controller reconnects automatically after reflash ;-)!