T-vK/ESP32-BLE-Keyboard #118: Send Android sleep

dominicklee, 2021-09-25:
Hi, thanks for the wonderful BLE keyboard library. I am trying to put my Android device to sleep using your BLE library.

Is there any keycode implementation that can sleep the screen of an Android device, specifically Android 5.1? I have tried many options but nothing works so far.

Hope you can help. Thanks.

T-vK, 2021-09-25:
I don't know, but that's out of scope for this library. You're probably better off asking in an Android forum for a keyboard shortcut to put a device to sleep.

dominicklee, 2021-09-26:
Okay. How about sending the "System Sleep" command, which is Usage ID 0x82 of the Generic Desktop Page (0x01) ?

How would I go about doing that?

T-vK, 2021-09-27:
You can try #3