T-vK/ESP32-BLE-Keyboard #119: How to set CFG_VENDOR_ID to prevent triggering keyboard assistant

acheong08, 2021-09-27:
Is there a way similar to digispark's `#define USB_CFG_VENDOR_ID 0xac, 0x05`

T-vK, 2021-09-27:
I don't know. I'm not even sure if there is a vendor id with Bluetooth besides the MAC address, device name and device manufacturer.

acheong08, 2021-09-28:
Any way to set MAC address to Apple's?

T-vK, 2021-09-28:
If that's possible you can probably find it somewhere on the Internet. The keyboard library by itself doesn't have that functionality.

sivar2311, 2021-09-28:
I have implemented functions in [my fork](https://github.com/sivar2311/ESP32-BLE-Keyboard) to set the vendor ID, product ID and version:
```C++
  void set_vendor_id(uint16_t vid);
  void set_product_id(uint16_t pid);
  void set_version(uint16_t version);
```

If you use values for the VendorID and ProductID that are [assigned to Apple](https://devicehunt.com/view/type/usb/vendor/05AC) it might work.

The values must be set before calling `bleKeyboard.begin()`:

```C++
void setup() {

  ...

  bleKeyboard.set_vendor_id(0x05ac); // Apple, Inc
  bleKeyboars.set_product_id(0x820a); // Bluetooth HID Keyboard
  bleKeyboard.begin();

  ...

}
```

closed 2022-05-15 by acheong08