T-vK/ESP32-BLE-Keyboard #120: ESP32 BLE Scan and BLE Keyboard

srudrawadayermath, 2021-09-29:
Hello,

My task is to scan a Bluetooth and send a message to the monitor whenever the scanned Bluetooth device is found. For this purpose my ESP32 should act as a keyboard. Also I have an LED interfaced, that blinks whenever the Bluetooth is detected.

I am able to scan and detect the bluetooth, and also blink the LED. But however when I tried to use BLEkeyboard, the loop isnot executed with the condition bleKeyboard.isConnected(). My questions are how do I know why this "if condition" is not satisfied and where can I read my bleKeyboard.print output? I have posted my code below, please help me with your inputs. Thank you.

#include <BleKeyboard.h>

/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

BleKeyboard bleKeyboard;

String Address = "58:ba:01:ef:ba:12";
const int Pin = 2;
int scanTime = 5; //In seconds
static BLEAddress *pServerAddress;
BLEScan* pBLEScan;

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks 
{
    void onResult(BLEAdvertisedDevice advertisedDevice) 
    {
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
      if (advertisedDevice.getAddress().equals(*pServerAddress))   
      {
        Serial.println("Device Match");                        
        digitalWrite (Pin, HIGH);                                 
        advertisedDevice.getScan()->stop();
      } // Found our server
      else
      {
        Serial.println("Device Mismatch");
        digitalWrite (Pin, LOW); 
      }
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");
  Serial.println("Starting BLE Keyboard");
  //Begin the BLE keyboard/start advertising the keyboard (so phones can find it)
  bleKeyboard.begin();
  
  pinMode(Pin, OUTPUT);
  pServerAddress = new BLEAddress(Address.c_str());
  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(100);
  pBLEScan->setWindow(99);  // less or equal setInterval value
}

void loop() {
  if (bleKeyboard.isConnected()) {
      Serial.println("Sending 'Hello world'...");
      bleKeyboard.print("Hello world");
      bleKeyboard.print(" ");
      pBLEScan->start(scanTime, false);
      pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory
      delay(2000);
  }
}

enthusisastSR, 2021-09-30:
Hi sivar2311,
Thank you for the response. I have not paired it yet but I have a switch, whose bluetooth is enabled when switched. For this purpose I need to hold it on. I am not sure how would I pair with it since I cannot read the detected devices on it and also unsure of how would I pair on ESP32.

sivar2311, 2021-09-30:
I am not yet sure exactly what your project is supposed to do. 
Can you describe it in a bit more detail?

I suspect there is a misunderstanding.
For example, what do you mean by "send a message to the monitor" ? Do you mean the serial monitor?

The KeyBoard BLE Library turns your ESP32 into a Bluetooth keyboard that can send keystrokes to a connected device (such as a PC). 
For a simple scan of Bluetooth devices and a display in the serial monitor you do not need this library.

enthusisastSR, 2021-09-30:
Sorry for the confusion. May be I am heading in the wrong direction. 

So I have an ESP32 board which is scanning this Bluetooth from a Switch (similar to Hypervolt switch). At present, the scanned device is displayed on the serial monitor and the led on my esp blinks each time its scanned.

Now, I want my ESP to send some data out through a serial interface like RS232 each time the bluetooth is scanned. Could you suggest me a better alternative than keyboard?

sivar2311, 2021-09-30:
```C++
#include <Arduino.h>
#include <BLEDevice.h>

const int  Pin      = 2;
BLEAddress serverAddress("58:ba:01:ef:ba:12");
BLEScan*   pBLEScan;

class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks {

    void onResult(BLEAdvertisedDevice advertisedDevice) {

        if (advertisedDevice.getAddress() == serverAddress) {

            Serial.printf("Device found! (%s)\r\n", serverAddress.toString().c_str());
            digitalWrite(Pin, HIGH); // turn on pin
            pBLEScan->stop(); // stop scanning

        } else {

            Serial.printf("Device Mismatch (%s)\r\n", advertisedDevice.getAddress().toString().c_str());

        }

        pBLEScan->clearResults();  // delete results fromBLEScan buffer to release memory
    }
};

void setup() {
    Serial.begin(115200);
    Serial.println("Scanning...");

//    pinMode(Pin, OUTPUT);
    BLEDevice::init("");
    pBLEScan = BLEDevice::getScan();  //create new scan
    pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
    pBLEScan->setActiveScan(true);  //active scan uses more power, but get results faster
    pBLEScan->setInterval(100);
    pBLEScan->setWindow(99);  // less or equal setInterval value
    pBLEScan->start(0, true); // scan until device found...
}

void loop() {}

```

enthusisastSR, 2021-09-30:
Thank you very much @sivar2311

closed 2021-09-30 by enthusisastSR