T-vK/ESP32-BLE-Keyboard #121: disconnects right away arch linux

Unknown-User2, 2021-09-30:
im on arch linux trying to do the example but when i connect it disconnects right away but it still tries to send the keys im using blueman for bluetooth

Unknown-User2, 2021-09-30:
i figured out it disconnects after it tries to send anything then reconnects

Unknown-User2, 2021-09-30:
doesnt work on windows either

sivar2311, 2021-09-30:
It works fine on windows:

https://user-images.githubusercontent.com/2558564/135385173-a0a6b089-c970-4743-aae8-67b46a4aa0b9.mp4

T-vK, 2021-09-30:
May it's a power supply issue? For example if you connect your ESP to a USB port on your computer it might not get enough current causing it to crash restart.

sivar2311, 2021-09-30:
My ESP is connected to the USB port of my computer. Otherwise I would not see the serial monitor output.
Have you tried another ESP module? Maybe it is faulty.
Anyway, I don't think the problem has to do with the library - it works fine.

Unknown-User2, 2021-09-30:
its showing when i go to look for bluetooth devices i can connect but when it tries to send anything like a key or hello world it disconnects

Unknown-User2, 2021-09-30:
i dont have another esp module
i am using the example but have also just tried one character
and im not sure which on i use

sivar2311, 2021-09-30:
Can you check the used ESP32 Arduino core version (Tools / Board  / Boards Manager...)
[image]

Unknown-User2, 2021-10-01:
i am using that but now im getting the error `lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)`

Unknown-User2, 2021-10-01:
i was getting that error with a different thing that was just sending 'A'

sivar2311, 2021-10-01:
`lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)` is triggered after disconnection is initiated. This error print mostly occurs in abrupt disconnection scenarios (example: connection timeout) but it is not the reason for disconnection.

It looks to me that the disconnection seems to come from the client (PC).
I guess your linux and windows is running on the same machine?
Do you have the same issues when connecting your mobilephone to the ESP?

Unknown-User2, 2021-10-01:
it connects fine to my iphone

sivar2311, 2021-10-01:
Does your linux and windows run on the same machine?
If it is so, i think it must have something to do with your machine hardware.

Unknown-User2, 2021-10-01:
there both the same machine

sivar2311, 2021-10-01:
Try on another machine

T-vK, 2021-10-01:
@Unknown-User2 Are you using a CSR Dongle by any chance? People have reported issues with those in the past. #37

@sivar2311 A regular USB port on a computer can theoretically supply 500mA of current which would be enough, but it's very common for that pool of 500mA to be spread over multiple ports, so if you have other devices using 400mA you could be in trouble already. I've also had a lot of issues with badly designed USB to serial adapters that weren't able to pass enough current through to the ESP in the past.

sivar2311, 2021-10-01:
@T-vK  Sorry, that was my fault. It was a misunderstanding. I had not paid attention to the sender of the comment and understood it as a question from Unknown-User2. (oops :man_facepalming:)

Since it connects to the smartphone and has been proven to work on both Linux and Windows, I suspect it's the hardware. The guess on a CSR dongle is good.

Unknown-User2, 2021-10-01:
im not using an adapter but i tried it on a linux mint machine it didnt work but it worked on a mac

T-vK, 2021-10-02:
Can you provide more details on your Bluetooth chip? Brand, model, Bluetooth version etc.