T-vK/ESP32-BLE-Keyboard #125: Add version number and add numpad keys

DustinWatts, 2021-11-09:
Hi T-vK!

Is it an idea to add version information to BleKeyboard.h? Something like this:
```
#define BLE_KEYBOARD_VERSION "0.0.3"
#define BLE_KEYBOARD_VERSION_MAJOR 0
#define BLE_KEYBOARD_VERSION_MINOR 0
#define BLE_KEYBOARD_VERSION_REVISION 3
```
This can help check for the version of the BleKeyboard during compile and warn if the programmer of the sketch wants to use a particular version.

Next, I included some definitions for the Numpad keys:

```
const uint8_t KEY_NUM_0 = 0xEA;
const uint8_t KEY_NUM_1 = 0xE1;
const uint8_t KEY_NUM_2 = 0xE2;
const uint8_t KEY_NUM_3 = 0xE3;
const uint8_t KEY_NUM_4 = 0xE4;
const uint8_t KEY_NUM_5 = 0xE5;
const uint8_t KEY_NUM_6 = 0xE6;
const uint8_t KEY_NUM_7 = 0xE7;
const uint8_t KEY_NUM_8 = 0xE8;
const uint8_t KEY_NUM_9 = 0xE9;
const uint8_t KEY_NUM_SLASH = 0xDC;
const uint8_t KEY_NUM_ASTERIX = 0xDD;
const uint8_t KEY_NUM_MINUS = 0xDE;
const uint8_t KEY_NUM_PLUS = 0xDF;
const uint8_t KEY_NUM_ENTER = 0xE0;
const uint8_t KEY_NUM_PERIOD = 0xEB;
```

Also, the definition for PRTSC is missing:

```
const uint8_t KEY_PRTSC = 0xCE;
```

T-vK, 2021-11-09:
Yeah, sounds great! Go ahead.

DustinWatts, 2021-11-09:
See https://github.com/T-vK/ESP32-BLE-Keyboard/pull/126