T-vK/ESP32-BLE-Keyboard #129: /BleKeyboard.h:8:34: fatal error: NimBLECharacteristic.h: No such file or directory

3Dietrich, 2021-11-14:
I tried NimBLE Keyboard by adding #define USE_NIMBLE before #include <BleKeyboard.h>
And it doesn't work. the error message is:
[..]/Arduino/libraries/ESP32-BLE-Keyboard/BleKeyboard.h:8:34: fatal error: NimBLECharacteristic.h: No such file or directory.
Did I a mistake?
I need the NimBLE version to get `isSubscribed()` working what [sends only `false`s in normal mode](https://github.com/T-vK/ESP32-BLE-Keyboard/issues/128#issuecomment-968280706)

sivar2311, 2021-11-14:
Please install NimBLE-Arduino library to use BleKeyboard in "NimBLE" mode.
This topic belongs to #128 and will be continued there.

closed 2021-11-14 by sivar2311

3Dietrich, 2021-11-14:
yes, got it. Thank you!