T-vK/ESP32-BLE-Keyboard #132: After the ESP32-C3 is powered off and auto-reconnected, the keys cannot be sent normally

Danjuanlab, 2021-12-23:
system ：windows
board: esp32-c3
esp firmware version: 2.01
blekeyboard version: V3.0 nimble

After connecting to the computer for the first time, the ble keyboard can work normally and maintain a stable connection. When the power supply of the development board is unplugged, and then reconnected, the development board will automatically reconnect with the original connected device (computer) and can be successfully connected. At this time, the key event can be triggered normally, but the key information cannot be received on the computer.
After I used the serial port to debug, I found the following:
```
[13:38:59.114]收←◆Sending f2
[ 49273][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 49273][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 49274][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 49281][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 49286][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 49292][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 49298][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 49304][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 49310][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 49316][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 49322][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 49328][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 49334][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 49340][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 49346][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 49352][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 49358][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 49364][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 49369][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 49376][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 49381][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 49388][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 49393][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 49400][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 49405][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 49411][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 49417][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 49423][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 49429][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 49435][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 49441][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 49447][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 49453][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 49459][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 49465][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 49471][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 49476][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 49483][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 49488][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 49495][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 49500][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 49507][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 49512][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 49518][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 49524][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 49530][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 49536][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 49542][D][BLECharacteristic.cpp:797] onStatus(): << onStatus

```

When the information can be received normally, the debug information is like this:
```
[13:39:27.358]收←◆Sending f2
[ 77517][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 77517][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 77519][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... Unknown
[ 77528][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 77534][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 77540][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 77546][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 77552][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... Unknown
[ 77561][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 77567][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 77573][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 77579][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 77586][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... Unknown
[ 77595][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 77601][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 77607][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 77613][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 77619][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... Unknown
[ 77628][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 77634][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 77640][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 77646][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 77652][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... Unknown
[ 77661][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 77667][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 77673][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 77679][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 77686][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... Unknown
[ 77695][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 77701][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 77707][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 77713][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
```
It can be seen that gattServerEventHandler() did not start correctly after power failure and reconnection

closed 2021-12-24 by Danjuanlab

pyIronManCoder, 2023-05-19:
Hello @Danjuanlab, I am also facing the same issue, it worked for the first time, later it doesn't. Could you please tell how you solved this issue.
Thank You

pyIronManCoder, 2023-05-21:
Ok  Thanks I found a solution from other issues. The solution was to use beta release version.