T-vK/ESP32-BLE-Keyboard #137: Enable iPhone onscreen keyboard when connected to esp32

shaddowgigs, 2022-01-11:
Is there a way to get the onscreen keyboard to pop up when i connect my esp32 as a bluetooth device? Im making a 5 key bluetooth keyboard for Switch Control on an iphone and if I set the keys to be used a regular keys the onscreen keyboard gets disabled. I tried using the media keys like play, next, volume etc and in that case the keyboard will show but Switch Control wont recognize it.

T-vK, 2022-01-13:
I don't have any Apple devices so can't really test this. I guess it's a feature of iOS to disable the software keyboard when a Bluetooth keyboard is connected?

tysonmatanich, 2022-01-25:
Changing the usage ([USB HID usage table](https://www.freebsddiary.org/APC/usb_hid_usages.php)) from Keyboard `0x06` to Keypad `0x07` does the trick for me. Is there a way to have this be configurable?

https://github.com/T-vK/ESP32-BLE-Keyboard/blob/183afacf04982af9961c6f47e64e2ed387f301ee/BleKeyboard.cpp#L34