T-vK/ESP32-BLE-Keyboard #138: Maybe no issue... ESP32 reset on begin with battery

ledsi, 2022-01-12:
Hi there,

I am not sure if its a library issue but my ESP32 restarts while I am calling begin in battery mode.

Booting my ESP32 works like a charm with TP4056 and 18650 battery connected.
ESP's 3V3 is provided by HT7333 and also seems to work fine.
But as soon as i call bleKeyboard.begin the ESP32 reboots.

Maybe its a powering issue because if i "jumpstart" the ESP32 circuit (TP4056 connected to 5V) it boots up and connects to Bluetooth device. After this i can disconnect TP4056 from 5V. It still works fine, no BT connection issue, no restart, nothing.
Everything works well.

Does anyone have an idea?
Is there an issue... bleKeyboard.begin is drawing too much current?

Thanks 4 help

T-vK, 2022-01-13:
Yeah this is not an issue with the library.
Calling the begin method most likely causes the radio to get turned on which results in a current spike that you need to handle. That HT7333 is probably not able to supply the amount of current that the ESP is trying to draw during that spike. Look up filter capacitors, a 10µF one would probably do the job. Or get a bigger regulator.

ledsi, 2022-01-13:
Hi @T-vK, thanks for this super fast reply.
I'm already wired up a 100uf tantalum and 100nf ceramic capacitor - but they cant handle it.

Maybe i really have to find a bigger voltage regulator, your right.
Do you have any suggestions?

cheers

T-vK, 2022-01-13:
Not really, I think I've successfully used [these boards](https://aliexpress.com/item/32885492501.html) in the past. They have a built in adjustable regulator.
There are also all in one solutions like [this one](https://de.aliexpress.com/item/1005001598299301.html).

ledsi, 2022-01-14:
Thanks a lot @T-vK and keep going on this great library!

closed 2022-01-14 by ledsi