T-vK/ESP32-BLE-Keyboard #141: Reconnecting Issue

gugo2808, 2022-01-22:
Hello Everyone. I have an android Samsung phone. 
The problem that when I reconnect my esp32 to my phone it won't control it any more. I have to delete connected device, reconnect again, and then it starts to control. I have tried both libraries 3.0.0 and 3.0.1

Thank you!

jandae, 2022-01-25:
I also encountered this issue on windows 10, thank you for letting us know which version works 

Cheers

bhoffmann84, 2022-02-07:
I am experiencing the same issue on Win11.
After power-off of either client or host they do reconnect but there is no functionality. Only re-pairing helps.

sivar2311, 2022-02-07:
I cannot confirm this.

The ESP32 automatically connects to my Windows 10 PC and my Samsung Galaxy A71 after a reboot.
I tested it with both standard BLE and NimBLE. 

I have to say that I am using PlatformIO, which still uses the older Arduino core (1.0.6) and is based on the ESP-IDF 4.3.2.

sivar2311, 2022-02-12:
I think we need to keep track of the different versions of each of the components used in order to make a comparison.

| Arduino Core | IDF-Version | BLE-Keyboard version | BLE-Keyboard mode | reconnect issue |
|---|---|---|---|---|
| 1.0.6 | 4.3.2 | 3.0.0 | classic | no |
| 1.0.6 | 4.3.2 | 3.0.0 | NimBLE | no |

Can you please also create a table for each of your tests?

I am going to setup ArduinoIDE to make a test on latest Arduino Core 2.x (or whatever is used right now)

sivar2311, 2022-02-12:
Forget about my last comment
It looks like the solution is much easier than I thought. :joy:

Solution: Use the latest version directly from the master branch.

All commits to fix this problem are still not present in the latest version, but they are present in the master branch!

Version 0.3.0 was released on 10 Sep 2021.
The fix was committed on 01 Oct 2021.

@T-vK Can you please publish the current master branch as a new release? This should avoid future problems of this kind.

thomasragos, 2022-02-13:
I can confirm that with the current master branch the issue is resolved.

Thanks @sivar2311!

T-vK, 2022-02-14:
I have released the current master as a new beta release now. 
Feel free to re-open in case this issue is not resolved yet.

closed 2022-02-14 by T-vK

gomcodoctor, 2022-11-11:
I am still facing same issue even with the latest beta version