T-vK/ESP32-BLE-Keyboard #143: Why is deviceName limited to 15 characters?

tysonmatanich, 2022-01-25:
Is this arbitrary or is there a reason to cap it at 15?

https://github.com/T-vK/ESP32-BLE-Keyboard/blob/183afacf04982af9961c6f47e64e2ed387f301ee/BleKeyboard.cpp#L99

I have connected with other devices that have supported names longer than 15 characters, and the [bluetooth docs](https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=229737) seem to align with this.

> 3.2.2 Bluetooth Device Name (the user-friendly name)
> 3.2.2.1 Definition
> The Bluetooth device name is the user-friendly name that a Bluetooth deviceexposes
> to remote devices. For a device supporting the BR/EDR device type,
> the name is a character string returned in the LMP_name_res in response to
> an LMP_name_req. For a device supporting the LE-only device type, the name
> is a character string held in the Device Name characteristic as defined in Section
> 12.1.

Or perhaps user-friendly name is a separate field?

> The Bluetooth device name can be up to 248 bytes (see [Vol. 2], Part C Section
> 4.3.5). It shall be encoded according to UTF-8 (i.e., name entered on the UI
> level may be down to 82 characters outside the Unicode range 0x00-0x7F are
> used).
> A device cannot expect that a general remote device is able to handle more
> than the first 40 characters of the Bluetooth device name. If a remote device
> has limited display capabilities, it may use only the first 20 characters.

tysonmatanich, 2022-01-25:
Also, calling `.setName(` before `.begin()` allows you to set a name longer than 15 characters. This is a great workaround but a strange behavior. It also first presents the user with the shorter name and then after connecting to the device displays the longer name. This seemed like an ok solution at first, however it seems to cause intermittent issues discovering the device initially. 😢

```
bleKeyboard.setName("01234567890123456789");
```

T-vK, 2022-01-25:
We added that limit because it caused massive issues for Apple users. When you set the device name so something longer than 15 characters it is supposed to get cut off and shortened to 15 chars automatically. If that only works through the constructor, but not through the setName method, I'd consider that a bug.