T-vK/ESP32-BLE-Keyboard #144: Hide Battery Level

Maxim0x00k, 2022-01-27:
Hi everyone! Can you say how can i hide battery level on my Iphone. I know what i can change battery level with (setbatterylevel) but if i choose 0% it still show thia
[image]

Maxim0x00k, 2022-01-27:
Thanks for the answer! Can you tell which lines are responsible for showing the battery ? Because if i don't use third paramiter (BleKeyboard bleKeyboard("Bluetooth Device Name", "Bluetooth Device Manufacturer", 100);) device shows me 100%. I guess that it is necessary to make changes to the "BleKeyboard.h" or "BleKeyboard.cpp"

T-vK, 2022-01-27:
I'd look for all lines saying something about battery.

closed 2022-02-04 by Maxim0x00k

BG2CRW, 2022-05-06:
you can set it 101(101~255), it will hide.