T-vK/ESP32-BLE-Keyboard #145: Error enabling NimBLE

pskerrett, 2022-02-06:
Hello,

I am having a problem compiling the example SendKeyStrokes with NimBLE enabled.

I am using v0.3.0 and also tried 0.3.1beta versions of the BleKeyboard libarary.

I enabled NimBLE by adding "#define USE_NIMBLE" to the first line of BleKeyboard.h

The error I am receiving:

```
In file included from \Arduino\libraries\ESP32-BLE-Keyboard\BleKeyboard.cpp:17:0:
\Arduino\libraries\ESP32-BLE-Keyboard\BleKeyboard.h:15:36: error: 'NimBLEHIDDevice' does not name a type
 #define BLEHIDDevice               NimBLEHIDDevice
                                    ^
\Arduino\libraries\ESP32-BLE-Keyboard\BleKeyboard.h:110:3: note: in expansion of macro 'BLEHIDDevice'
   BLEHIDDevice* hid;
   ^
\Arduino\libraries\ESP32-BLE-Keyboard\BleKeyboard.cpp: In constructor 'BleKeyboard::BleKeyboard(std::__cxx11::string, std::__cxx11::string, uint8_t)':
\Arduino\libraries\ESP32-BLE-Keyboard\BleKeyboard.cpp:98:7: error: class 'BleKeyboard' does not have any field named 'hid'
     : hid(0)
       ^

```

The script compiles & functions properly when I do not add USE_NIMBLE to the BleKeyboard.h file... I'm attempting to use NimBLE to reduce memory consumption.

Thanks!

sivar2311, 2022-02-06:
Unfortunately there is a small bug in the BleKeyboard.cpp.
Please edit the BleKeyboard.cpp file. 
Move `#include "BleKeyboard.h"` from line 17 to line 1

pskerrett, 2022-02-06:
Perfect! That works. Thank you. I see a reduction in memory now!

closed 2022-02-06 by pskerrett

sivar2311, 2022-02-06:
Thank you for reminding me.
I have now fixed this error in the master branch.