T-vK/ESP32-BLE-Keyboard #147: How to put ESP32 in 'pairing mode'

joostbijl, 2022-02-19:
Hi,

i have created a set of handlebar buttons for navigating on my motorcycle, see details here: https://jaxeadv.com/gear-gadgets/navigation-setup/handlebar-buttons/. This library is really useful for that, i'm a big fan!

What i can't seem to work out is how to pair the esp32 to a new phone. I'm looking for the way to put the esp32 in pairing mode so that i can connect it to a different phone (forgetting the previously connected phone).

I'm assuming it has to do with the BLEScan method?  But i'm probably missing something, because all the info i can find seems very convoluted for the simple thing i'm trying to do.

Thanks!

sivar2311, 2022-02-19:
Hi @joostbijl !
That is a very nice project! 👍 

I'm not a BLE expert and maybe I don't understand the question correctly, but the BleKeyboard is automatically in pairing mode unless there is a connection.

Have you tried the BleKeyboard in NimBLE mode as well?

joostbijl, 2022-02-19:
Hi @sivar2311! Thanks, it's really useful to control your phone with your motorcycle gloves on ;)

It does indeed accept new connections when it's not connected!.

Now i have the issue that i want to remove the other phone i connected it to. (or: just remove every existing connection).

Earlier today i used NimBLE, but then the esp would connect, but not send actual keystrokes. I reverted back to the original library and now it works. Did not investigate further. Would it help you if i did?

sivar2311, 2022-02-19:
Please download the latest [beta version](https://github.com/T-vK/ESP32-BLE-Keyboard/releases) and activate the NimBLE-Mode (uncomment the 2nd line in [BleKeyboard.h](https://github.com/T-vK/ESP32-BLE-Keyboard/blob/f8dd4852113a722a6b8dc8af987e94cf84d73ad5/BleKeyboard.h#L2))

There are a lot of changes since the last official release. I think this will fix your issues.

joostbijl, 2022-02-19:
NimBLE works, after i have removed the esp from my phone and reconnected.

sivar2311, 2022-02-19:
Does it mean the latest beta solved the issue?

joostbijl, 2022-02-19:
The original issue is not solved, but not really necessary since you can 'forget' devices from your phone just as easily.

Using NimBLE i can now run the ble_store_clear() command, but that does not appear to clear all bonds stored on the esp.

Guillemete, 2022-04-05:
Maybe you can reset the internal memory on the ESP where the pairing info if stored?

Guillemete, 2022-04-05:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/bt_remove_paired_devices/bt_remove_paired_devices.ino

this might help. havent tried it yet

sivar2311, 2022-04-05:
NimBLE supports this:
```NimBLEDevice::deleteAllBonds();```

But I can't say how it is done with the standard BLE library.