T-vK/ESP32-BLE-Keyboard #149: Mac OS cannot connect the device

BeijingUncle, 2022-03-01:
Try the example code with a Mac mini, when click connect, got:
E (8531) BT_GATT: GATT_INSUF_AUTHENTICATION
then few seconds later:
E (41608) BT_BTM: Device not found
OS: Monterey 12.2.1.

Thanks

T-vK, 2022-03-01:
Try using the latest beta release.

dejavu1987, 2023-03-26:
> Try using the latest beta release.

Just tried with 0.3.2-beta but same issue.
```
E (25277) BT_GATT: GATT_INSUF_AUTHENTICATION

E (25627) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
E (25628) BT_SMP: smp_derive_link_key_from_long_term_key failed

E (25634) BT_BTM: btm_proc_smp_cback received for unknown device
E (25638) BT_BTM: BTM_GetSecurityFlags false

E (25642) BT_GATT: GATT_INSUF_AUTHENTICATION

E (25761) BT_BTM: BTM_GetSecurityFlags false

E (25761) BT_GATT: GATT_INSUF_AUTHENTICATION

E (25783) BT_BTM: BTM_GetSecurityFlags false

E (25784) BT_GATT: GATT_INSUF_AUTHENTICATION

Sending 'Hello world'...
Sending Enter key...
Sending Play/Pause media key...
E (27966) BT_BTM: BTM_GetSecurityFlags false

Waiting 5 seconds...
E (29068) BT_BTM: BTM_GetSecurityFlags false

Sending 'Hello world'...
Sending Enter key...
E (35132) BT_BTM: BTM_GetSecurityFlags false

E (35166) BT_BTM: BTM_GetSecurityFlags false

E (35233) BT_BTM: BTM_GetSecurityFlags false

E (35256) BT_BTM: BTM_GetSecurityFlags false
```

dejavu1987, 2023-03-26:
I have tested this with ESP32-S3 and ESP32 and issue is only there for ESP32-S3, is it supported at all?