T-vK/ESP32-BLE-Keyboard #151: This works only with usb?

jorgeluisrezende, 2022-03-07:
My question is, can i use this firmware to create an usb keyboard, can i use this library for it?

sivar2311, 2022-03-07:
This library is for Bluetooth Low Energy only.

For a USB keyboard you need an ESP32-S2 chip, as only this has the required USB functionality.

T-vK, 2022-03-07:
I've used the standard Arduino usb keyboard library on an Arduino Leonardo and on an Arduino Micro before. It's simple and works really well. 
I have done USB keyboard stuff on an ESP8266 before using the usb code provided by cnlohr. But it's really hacky and tough to set up.

jorgeluisrezende, 2022-03-07:
@T-vK nice, this library has suport for layering? I want to make a 40% keyboard

T-vK, 2022-03-08:
I don't know which library you mean or what you mean by layering.

jorgeluisrezende, 2022-03-08:
@T-vK sorry, I was not clear. I ask if [ESP32-BLE-Keyboard](https://github.com/T-vK/ESP32-BLE-Keyboard) has layering suport to create keys combination directly on keyboard like gmk fimeware. This is something that i need to have to make a 40% keyboard. Other question when you use ESP8266 to make a keyboard which fimeware did you use on it to make all the stuf work?

T-vK, 2022-03-08:
I still don't know what you mean by layering. I also don't know what the gnk firmware is. 
I'm simply providing a bluetooth keyboard library that's supposed to be used with the Arduino IDE. I don't know what the firmware is called, but this is all managed by the IDE.
https://github.com/T-vK/ESP32-BLE-Keyboard/blob/master/README.md

jorgeluisrezende, 2022-03-08:
Other question when you use ESP8266 to make a keyboard which fimeware did you use on it to make all the stuf work?

sivar2311, 2022-03-08:
Maybe I'm misunderstanding, but to clear things up:
- This is a library for the Arduino Framework for the ESP32.
- The Arduino Framework is used to program your own "custom" firmwares. (Every Arduino sketch results in a "firmware")

jorgeluisrezende, 2022-03-08:
@sivar2311 Right, one more question, as you said this library is for esp32 but  will it works on esp8266? if not, do you know some lib that I can use on esp8266 to make a keyboard?

jorgeluisrezende, 2022-03-08:
@sivar2311 Right, I already look at that driver, thank you and @T-vK  for attention!

closed 2022-03-08 by jorgeluisrezende