T-vK/ESP32-BLE-Keyboard #152: esp32s3 authentication errors

bm16ton, 2022-03-27:
Hello,
   Im new to this but the lib/sketch was unable to authenticate correctly on esp32s3, im foolish but i think it requires dual mode and S3 is strictly BLE no BR, but the mouse lib worked so i copied the auth part overe and all is good. So in short changed

  BLESecurity* pSecurity = new BLESecurity();
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND);

to

  BLESecurity *pSecurity = new BLESecurity();
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);

taken from the working mouse lib

DustinWatts, 2022-05-11:
May I suggest a Pull request? 

Maybe using something like:

`if CONFIG_IDF_TARGET_ESP32S3` ?

bm16ton, 2022-05-12:
oohh never realy done one of those! Ha its the little things in life isnt
it!? hopefully will get to it within next couple days! thank you for the
suggestion.

Hannah-GBS, 2022-07-15:
esp32-c3 needs the same fix

dejavu1987, 2023-04-05:
So this is what I was looking for a week now? :)

bm16ton, 2023-04-05:
crap, never use this so totaly forgot, i realy should do a pr. Glad it
wirked for you tho

dejavu1987, 2023-04-24:
Actually I switched to NimBLE and it is apparently doing the same.

bm16ton, 2023-04-24:
? I guess i dont understand. If you try and use the function/protocol that
needs both "legacy" and ble bluetooth versions to function on the S3  which
only supports ble, it wont work on any lib etc. Is this what happened? Or
something else? Rereading this it sounds like im being pissy but im not,
read it while picturing me smiling trying to be helpful. im bad at written
word for communicating.