T-vK/ESP32-BLE-Keyboard #156: Not Working On ESP32-S3 WROOM

TANG617, 2022-04-22:
I am using the following ESP32-s3 library on PlatformIO
https://github.com/tasmota/platform-espressif32/
However, even the example doesn't works fine on my ESP32-S3, with no error or warning.
I tried to upload the exactly same code to ESP32(not S3), it worked perfectly.
I tried every mean, but got myself in desperation, Please Help Me!!!!!!!!!
The Log was posted as follow(No Error, No Warning, Seem to succeed, But doesn't work)
```bash
> Executing task: platformio run --target upload --environment esp32s3 <

Processing esp32s3 (platform: https://github.com/tasmota/platform-espressif32/archive/refs/heads/IDF44/ESP32-S3.zip; board: esp32-s3-devkitc-1; framework: arduino)
---------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
FILESYSTEM  spiffs
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (2.0.3) > Espressif ESP32-S3-DevKitC-1
HARDWARE: ESP32S3 240MHz, 512KB RAM, 8MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 2.0.3 
 - tool-esptoolpy 1.30201.220126 (3.2.1) 
 - tool-mklittlefs 1.203.210628 (2.3) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51) 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s3 8.4.0+2021r2-patch2
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32 BLE Keyboard> 0.3.2
|   |-- <ESP32 BLE Arduino> 2.0.0
Building in release mode
Retrieving maximum program size .pio/build/esp32s3/firmware.elf
Checking size .pio/build/esp32s3/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.3% (used 27964 bytes from 524288 bytes)
Flash: [====      ]  36.1% (used 472781 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.wchusbserial10
Uploading .pio/build/esp32s3/firmware.bin
esptool.py v3.2.1
Serial port /dev/cu.wchusbserial10
Connecting..........................
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:41:c8:34
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00083fff...
Compressed 12592 bytes to 8871...
Writing at 0x00000000... (100 %)
Wrote 12592 bytes (8871 compressed) at 0x00000000 in 0.5 seconds (effective 190.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 223.7 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.2 seconds (effective 332.7 kbit/s)...
Hash of data verified.
Compressed 473168 bytes to 329265...
Writing at 0x00010000... (4 %)
Writing at 0x0001b731... (9 %)
Writing at 0x00020d59... (14 %)
Writing at 0x0002672f... (19 %)
Writing at 0x0002bbad... (23 %)
Writing at 0x000312cf... (28 %)
Writing at 0x00036bd8... (33 %)
Writing at 0x0003be26... (38 %)
Writing at 0x00040e86... (42 %)
Writing at 0x00045db8... (47 %)
Writing at 0x0004bbc9... (52 %)
Writing at 0x0005160f... (57 %)
Writing at 0x000569eb... (61 %)
Writing at 0x0005bedd... (66 %)
Writing at 0x0006177d... (71 %)
Writing at 0x00066fc5... (76 %)
Writing at 0x0006c849... (80 %)
Writing at 0x000724ac... (85 %)
Writing at 0x00077515... (90 %)
Writing at 0x0007d096... (95 %)
Writing at 0x00082d0e... (100 %)
Wrote 473168 bytes (329265 compressed) at 0x00010000 in 8.8 seconds (effective 428.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
=============================== [SUCCESS] Took 18.93 seconds ===============================

Environment    Status    Duration
-------------  --------  ------------
esp32s3        SUCCESS   00:00:18.927
================================ 1 succeeded in 00:00:18.927 ================================
```

T-vK, 2022-04-22:
I don't even have an ESP32-S3 to test this, but I would guess that there isn't much I can do if the that ESP is too different architecturally so that the Bluetooth library would have to be rewritten.

TANG617, 2022-04-24:
Thanks for your reply! So, the problem lies in the official arduino BLE library of ESP32-S3 and I put in my hope to it could be solved with further official updated.
Ps, there are few ESP32S3 remaining and if it's helpful, I would like to give away some of them to you!

bm16ton, 2022-04-29:
The only difference between the two is the s3 doesnt have classic bluetooth so authentication was failing because the method used required dual mode support (even tho was using ble) see previous issue, https://github.com/T-vK/ESP32-BLE-Keyboard/issues/152. Im not sure what error messages you where getting. But does this seem to be related? Does the mouse lib work?

TANG617, 2022-05-08:
It WORKS !!!!!
Thanks so much!
You do save my days!

bm16ton, 2022-05-08:
Oh no problem super glad I could help! I killed my S3 by accidentally hitting a peizo buzzer while it was attached and the generated voltage killed it. I only mention it cuz hands down the weirdest, coolest way ive ever killed a mcu! lol

TANG617, 2022-05-13:
> Oh no problem super glad I could help! I killed my S3 by accidentally hitting a peizo buzzer while it was attached and the generated voltage killed it. I only mention it cuz hands down the weirdest, coolest way ive ever killed a mcu! lol

Yesterday I killed my RaspberryPi4 with a tiny metal scissor accidentally. I am sure it's a much more stupid way to kill a mcu(if raspberrypi counts). my wallet is bleeding...

bm16ton, 2022-05-13:
Ha! my god the price of raspberrypi 4s has skyrocketed (or so i was told).
I bought a hackrf back when they was brandnew, opened the box started it,
held it close to my laptops wifi hit a button in sum app and she was fried.
$500 in less then one minute. Christ i have a lot of stories killing
electronics. Think the Pi maybe salvagable?

TANG617, 2022-05-13:
LOL, Thanks for sharing the story of MURDERING electronics. After my carelessness on Pi, I do have repaired it! It's not easy and very risking. Maybe I can help you diagonse where the problem lies. I have always complained about how electronics have merely no protection of stupidity. There's always a frustrating moment after a wisp of smoke rising from the expensive chips.

LIUHanyang96, 2022-08-30:
Hi, I am using WROOM 32 and trying to follow #152 , but my WROOM32 still cannot pair to my PC or phone.