T-vK/ESP32-BLE-Keyboard #159: Not working on reconnection

JoDe52078, 2022-05-01:
BleKeyboard is working fine after first pairing with an Android device, but no response after reconnection.
I tried this with three different phones/tablets.
BleKeyboard in recognized and reconnected immediatly after power on.
I always have to remove the keypad from the Android device and connect as new to get it working again.
Any help appreciated.

sivar2311, 2022-05-01:
Please try the [0.3.2 beta release](https://github.com/T-vK/ESP32-BLE-Keyboard/releases/tag/0.3.2-beta).
This issue should be fixed in the beta version.

JoDe52078, 2022-05-01:
> 
Wow, I'm impressed!
It works fine now.
Very fast responce and the solution!
Thank you so much!

badmaxx, 2022-05-05:
Had the same issue with Windows 10, using the beta fixed the problem.

vcchm, 2022-10-08:
Yes but beta does not connect to Windows 11 ou new Android (Samsung Galaxy note 20)...

mayaku2go, 2023-01-21:
Here's the solution:
https://github.com/T-vK/ESP32-BLE-Keyboard/issues/168#issuecomment-1399208029