T-vK/ESP32-BLE-Keyboard #160: Call management with library

vpeinadolopez, 2022-05-06:
Hi, I'm using the library to control music or videos on Android and iOS reliably. The next step I want is to be able to pick up calls and hung them up. I've tried every media commands and most of the others, the only thing I achieved is picking up calls only on Android with KEY_MEDIA_PLAY_PAUSE, but then the command just mutes the microphone, it doesnt hang up.

I'd really appreciate some help to achieve this or at least a response if it's not possible to do so.

T-vK, 2022-05-06:
I don't know. Your question is basically if there is a key or keyboard shortcut that can hang up calls on Android. That seems like a question to ask on an Android forum.

vpeinadolopez, 2022-05-10:
Thanks for your response.
Just to see if I get a clue. Where did you get the commands from? For example that KEY_ESC = 0xB1 or KEY_MEDIA_STOP = {4, 0}. I'd like to do some research to see if there's any specific command for what I need.

T-vK, 2022-05-17:
It's from the USB spec. Not sure which document it was. It's always difficult to find, but it's somewhere on https://www.usb.org/

vpeinadolopez, 2022-05-23:
Yes it is: [](https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf)
Now i finally discovered a way to hang up calls on Android, if you press twice KEY_MEDIA_PLAY_PAUSE with a delay between 500 and 1000 miliseconds and then release, it works. But only on Android and I dont feel like it´s the most elegant way.
I'm trying to make a new collection in _hidReportDescriptor using the Telephony Device Page (0x0B) in USB spec and command Answer On/Off (0x74), but I'd need some help to get it fully working, specially with the collection parameters.