T-vK/ESP32-BLE-Keyboard #161: `sdkconfig.h: No such file or directory` on a fresh install for Arduino Nano

beyarkay, 2022-05-07:
Hi there, I'm getting the error:
```
Detecting libraries used...
[omitted long line]
Alternatives for BleKeyboard.h: [ESP32-BLE-Keyboard@0.3.0]
ResolveLibrary(BleKeyboard.h)
  -> candidates: [ESP32-BLE-Keyboard@0.3.0]
[omitted long line]
Alternatives for sdkconfig.h: []
ResolveLibrary(sdkconfig.h)
  -> candidates: []
In file included from /<homedir>/Documents/Arduino/libraries/ESP32-BLE-Keyboard/examples/SendKeyStrokes/SendKeyStrokes.ino:4:0:
/<homedir>/Documents/Arduino/libraries/ESP32-BLE-Keyboard/BleKeyboard.h:3:10: fatal error: sdkconfig.h: No such file or directory
 #include "sdkconfig.h"
          ^~~~~~~~~~~~~
compilation terminated.
Using library ESP32-BLE-Keyboard at version 0.3.0 in folder: /<homedir>/Documents/Arduino/libraries/ESP32-BLE-Keyboard 
exit status 1
Error compiling for board Arduino Nano 33 BLE.
```

I'm hoping to get this working for the Arduino Nano 33 BLE. Is this possible?

sivar2311, 2022-05-07:
Hello @beyarkay 

This library is written for the ESP32 chipset and cannot be used with the Arduino Nano (ATmega328).

beyarkay, 2022-05-09:
Thanks for the fast reply!

Is there no way I can modify it to work with the Nano? I'm happy to put in the work and contribute but my use case requires out be compatible with the Nano

sivar2311, 2022-05-09:
No, this is not possible, because this library depends on the functions of the ESP32 chip. 
You either need a library for the Arduino Nano BLE like [mbed-ble-hid](https://github.com/tcoppex/mbed-ble-hid) or you could alternatively use an ESP32 devkit or similar.