T-vK/ESP32-BLE-Keyboard #163: Bluetooth keyboard

JohnTaipei, 2022-05-31:
Hello Sir:
use your library to make a Bluetooth keyboard
https://www.facebook.com/100044063329321/videos/355510663344838/
Thank for your library

creepscompilation, 2022-06-03:
Hi John, 
Your project is awesome, looks GREAT !!!

I am NEW to using the ESP32 so I was hoping you might be able to point me in the right direction?

I have a similar application that requires the ESP32 to communicate with another device via a serial UART and
I want to create a series of UART inline commands within the ESP32 that the other device can use.

I will need to be able to command the ESP32 BLE keyboard module to:
- REBOOT or RESET
- Start Over or RESET settings, waiting to pair...

I already have a basic setup working where I can send keyboard codes via the Arduino Serial Monitor Window.

Problem: 
Android connects but does not display any characters, all other operating systems work OK.
I have tried loading in the latest library, but I am not sure how to know if the latest library is actually installed
and being used?

Also, the Android does not remember the connection and wants to PAIR every time?
Apple iPad, Windows 8, Windows 10, all work OK...

Latest library I can find:
https://github.com/T-vK/ESP32-BLE-Keyboard/releases/tag/0.3.2-beta

Thank you in advance for any suggestions.

Mike Balistreri
Irvine, Ca.

JohnTaipei, 2022-06-04:
@creepscompilation 
Thanks

problem 1: the same as you
problem 2: the same as you

i am not good at software
you can ask him who also lives Irvine, Ca.
https://www.youtube.com/c/ThatProject

creepscompilation, 2022-06-10:
Hi John,

Can you let me know if find a solution to the Android failures?