T-vK/ESP32-BLE-Keyboard #165: Updating version?

creepscompilation, 2022-06-08:
I have not been able to get Android devices to work, Windows and Apple MacOS, iOS work well.

I read that I should update the version, so I downloaded the latest version and added the library ZIP,
however, Android still does not work. 

Android connects, but no keystrokes are received.
Android seems to "pair" every time and never shows up under the "paired" device list.

I cannot find this ESP32-BLE Keyboard listed in the library manager to confirm the current version.
Question: How do I confirm the version number and correctly update the version?

I apologize if this seems like simple question, but all other libraries I have installed show up in the library
manager and I am able to confirm the version being used and install new versions.

SunandMittal, 2022-06-08:
Dear @creepscompilation
I had similar observations in Android and possible workaround, which I just posted  #166 just now.. 
You can give it a try and see if it works for you.

creepscompilation, 2022-06-09:
Ok, I managed to figure out how to remove the ESP32-BLE-Library and then install a different version.
I have tried the following:
v0.3.2beta
v0.3.1beta
v0.3.0
Works on all operating systems except Android for some reason.