T-vK/ESP32-BLE-Keyboard #167: Everything works except for Android..

creepscompilation, 2022-06-10:
Hello,

I have fully tested the following version of the ESP32-BLE-Keyboard library using an ESP32-S3 module
and everything works GREAT, except for Android..

v0.2.3
v0.3.0
v0.3.1
v0.3.2

I confirmed the library versions prior to testing by removing the library completely, closing and opening the Arduino IDE, and
installing the ESP32-BLE-Keyboard library clean.

Please let me know if I am doing something wrong here?

I am hoping that I am just doing something stupid, I am new to the Arduino IDE.

makerbitter, 2022-06-13:
I have the same problem in some Android phones. The esp32 connect correctly but the keyboard keystrokes are not interpreted by the android phone

dailafing, 2022-07-15:
I can confirm that I have the same problem.
Used the Example file, the ESP32 is connecting and paired, but no keystrokes are being interpreted by Android.
No idea why....

`#include <BleKeyboard.h>

BleKeyboard bleKeyboard;

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleKeyboard.begin();
}

void loop() {
  if(bleKeyboard.isConnected()) {
    Serial.println("Sending 'Hello world'...");
    bleKeyboard.print("Hello world");

    delay(1000);
    bleKeyboard.releaseAll();
  }

  Serial.println("Waiting 5 seconds...");
  delay(5000);
}`

bertenvdb, 2022-08-28:
Same here with v0.3.2 on Android 12.

eburrows, 2022-09-05:
I have a similar (maybe same) problem: After initial paring, my ESP32 sending media keys (volume/next/prev) works fine, but after a power cycle, the keys are not read by my Android phone, though the device shows as connected in the bluetooth settings screen.

To verify if you have the same issue, forget and re-pair the device, and see if the keys come through. This works for me, though I would like a solution!

sipamon, 2026-03-14:
Has a solution been found in the meantime? I have the same problem with my Samsung tablet; it doesn't work there. But it does work on my OnePlus phone. I really don't know why.