T-vK/ESP32-BLE-Keyboard #168: Won't re-connect on Windows 10 after Power Cycle Off-On Release v0.3.0

creepscompilation, 2022-06-15:
I am in the final phase of testing and I have to use release v0.3.0 to get Android to work at all..

So, with release v0.3.0, Android and Apple iOS work GREAT !!!
On Windows 10 however, I can PAIR and Connect OK, and the BLE Keyboard works GREAT, until I reset
the ESP32 module either power off-on, or boot button.
After that, "isConnected" comes up as connected, but no keyboard data outputs to Windows 10.

I am forced to UNPAIR the BLE keyboard from Windows 10, the PAIR again.
This test is repeatable and confirmed.
This probem does not exist with Android or Apple iOS, only Windows 10.

creepscompilation, 2022-06-15:
Correction: Won't re-connect on Windows 8 also..
Same issue with Windows 10, Windows 8

creepscompilation, 2022-06-15:
RESOLVED:
I took the advice from SunandMittal..

Starting with release v0.3.2
filename = "BleKeyboard.cpp"

Find this line:
pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND);

Replace the above line to the following from release v0.3.0
pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);

So, far.. 
Android connects, re-connects, keyboard codes work
Apple iOS connects, re-connects, keyboard codes work
Windows 10 connects, re-connects, keyboard codes work

creepscompilation, 2022-07-01:
Hello mvermand22,

I started with v0.3.2, then I changed that line in the code, I did not test this on any earlier releases.
I have attached the library that I modified, NOTE: I call it v0.3.3 for my own reference, this is NOT an official
release from this GIT HUB page.

[ESP32-BLE-Keyboard 0.3.3.zip](https://github.com/T-vK/ESP32-BLE-Keyboard/files/9030095/ESP32-BLE-Keyboard.0.3.3.zip)

jnsw, 2023-01-17:
> Hello mvermand22,
>
> I started with v0.3.2, then I changed that line in the code, I did not test this on any earlier releases. I have attached the library that I modified, NOTE: I call it v0.3.3 for [...]

thanks, this worked for me!

ekoslav, 2023-01-18:
On Windows 11 issue still remains. On the first connection the keyboard works, however if it disconnects, and connects back, no keys are sent. Does anyone know any tool that will log the messages from BT devices on Win 11? I check the system logs, but they only show when the device was first paired.

mayaku2go, 2023-01-21:
I had the same problem and created just an account for this.
Here is the solution:
https://github.com/nkolban/esp32-snippets/issues/632

Open bleKeyboard.cpp and add the two lines  to the begin function.
```
uint8_t val[] = {0x01, 0x00};
desc->setValue(val, 2);
```

The whole function should look like this:
```
void BleKeyboard::begin(void)
{
  BLEDevice::init(deviceName);
  BLEServer* pServer = BLEDevice::createServer();
  pServer->setCallbacks(this);

  hid = new BLEHIDDevice(pServer);
  inputKeyboard = hid->inputReport(KEYBOARD_ID);  // <-- input REPORTID from report map
  outputKeyboard = hid->outputReport(KEYBOARD_ID);
  inputMediaKeys = hid->inputReport(MEDIA_KEYS_ID);

  outputKeyboard->setCallbacks(this);

  hid->manufacturer()->setValue(deviceManufacturer);

  hid->pnp(0x02, vid, pid, version);
  hid->hidInfo(0x00, 0x01);

#if defined(USE_NIMBLE)

  BLEDevice::setSecurityAuth(true, true, true);

#else

  BLESecurity* pSecurity = new BLESecurity();
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND);

#endif // USE_NIMBLE

  hid->reportMap((uint8_t*)_hidReportDescriptor, sizeof(_hidReportDescriptor));
  hid->startServices();

  onStarted(pServer);

  advertising = pServer->getAdvertising();
  advertising->setAppearance(HID_KEYBOARD);
  advertising->addServiceUUID(hid->hidService()->getUUID());
  advertising->setScanResponse(false);
  advertising->start();
  hid->setBatteryLevel(batteryLevel);

  ESP_LOGD(LOG_TAG, "Advertising started!");
}
```

vcchm, 2023-01-21:
@mayaku2go : you did not add the lines you recommend to add. Where should they be ? At the beginning ? At the end ?

mayaku2go, 2023-01-21:
@vcchm My fault. Was the wrong function. I updated the code ^^

vcchm, 2023-01-22:
Thanks ! Your correction only changes the code in the "Nimble" mode. But the problem happens whatever the mode - Nimble or not. Are you sure you changed the right code ?

vcchm, 2023-01-22:
Also are you sur you dont need 
    desc->setValue(val, 2);
after the first desc->setNotifications(true); too ???

mayaku2go, 2023-01-22:
> 

Haven't tested it yet as I still don't understand BLE completely XD
Maybe one of the developers are able to answer this question. 
It worked for me and I am trying to establish a secure connection with no input or output and static passkey right now.
But this is a point worth trying.

mayaku2go, 2023-01-25:
@vcchm You are right. Without the second desc->setValue(val, 2); the ESP32 isn't recognized as an HID Keyboard. Code is updated ^^