T-vK/ESP32-BLE-Keyboard #169: Solution seems to Work..

creepscompilation, 2022-06-15:
RESOLVED:
I took the advice from SunandMittal..

Starting with release v0.3.2
filename = "BleKeyboard.cpp"

Find this line:
pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND);

Replace the above line to the following from release v0.3.0
pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);

So, far..
Android connects, re-connects, keyboard codes work
Apple iOS connects, re-connects, keyboard codes work
Windows 10 connects, re-connects, keyboard codes work

SunandMittal, 2022-06-16:
Dear @creepscompilation, 
good to hear that workaround solution worked for you :-).

regards,
Sunand

makerbitter, 2022-06-20:
Hello @SunandMittal @creepscompilation I'm using ninBLE option. How could I solve this? I need use nimBLE option activated. I tested in android, connect and reconnect but keyboard codes don't work.

SLboat, 2023-01-07:
it's work for me too!!!

justinkrutz, 2023-01-15:
This worked for me as well on my OnePlus 9 with Android 13

SirBramble, 2023-02-20:
it worked for me too!!!

Cengls, 2023-06-16:
Holy god Man !!!!!!
WORKING!!!!!!!!!!!!!! 
 IN 
WINDOWS11 
ANDROID10

creepscompilation, 2023-06-16:
> 

Hey Cengls, 
I'm happy to help..
I wish I could take credit, but I did alot of Google searches until I
was able to find that solution.
Keep bangin' on it..

Cengls, 2023-06-16:
> Hey Cengls,
> I'm happy to help..
> I wish I could take credit, but I did alot of Google searches until I
> was able to find that solution.
> Keep bangin' on it..

Ur Best ：D