T-vK/ESP32-BLE-Keyboard #171: print() method do not send more than 16 chars per call.

aopenna, 2022-06-24:
Hello,

I'm working on a prototype that read from a HID USB device and send data as a bluetooth keyboard. I use the USB host library to read data from the USB device, and i have a custom HID parser to read the data. This is working flawlessly. 

than I added the BLe keyboard, and to send data as the BLE keyboard I write the data processed by custom parser on an char array, and set a flag that will be read on loop (as below)

The problem is if the array has more than 16 chars the paried device only received the first 16, some time 20, but nothing more. As an workaround I send in 16 chars chunks until the end of array. Even using this workaround I needed to put a 100 ms delay in order to not lost any chars.

Anyone have the similar problem?

```cpp
void loop()
{
  if (sendCharstoBLE) {
    sendCharstoBLE = false;
    sentToBLE = true;
    Notify(PSTR("\r\nSent to BLE\r\n"), 0x80);
    if (bleKeyboard.isConnected()) { //send to BLE
      Notify(PSTR(" BT connected, sending..."), 0x80);
      int i;
      String charSubstring;
      for ( i = 0; i <= charstoBLE.length(); i += 16) {
        Notify(Serial.print(charSubstring),0x80);
        Notify(PSTR("\r\n"),0x80);
        Notify(PSTR(charSubstring.length()),0x80);
        charSubstring = charstoBLE.substring(i, 16);
        bleKeyboard.print(charSubstring);
        delay(100);
      }
      charSubstring = charstoBLE.substring(i);
      if (charSubstring.length() > 0) {
        Notify(Serial.print(charSubstring),0x80);
        Notify(PSTR("\r\n"),0x80);
        Notify(PSTR(charSubstring.length()),0x80);
        bleKeyboard.print(charSubstring);
        delay(300);
      }
      //bleKeyboard.print(charstoBLE // send all chars is not working, disabling for while
      delay(300);
      bleKeyboard.write(KEY_RETURN);
      delay(100);
    } else {
      Notify(PSTR(" BT disconnected"), 0x80);
    };
    delay(300);
  }
  Usb.Task();
}
```

ekoslav, 2022-07-13:
I had that issue, where the chars would drop randomly or lock up repeating character indefinitely on long block (like an e-mail address)
I resolved the issue by sending chars one by one. Now I can send unlimited length, and I can control the speed of typing.

alexz006, 2023-01-31:
```
void Keyboard_print(String text){
  if(text.length()>15){
    Keyboard.print(text.substring(0,15));
    text.remove(0, 15);
    Serial.println(text);
    delay(100);
    return Keyboard_print(text);
  }
  Keyboard.print(text);
}

Keyboard_print("Lorem ipsum dolor sit amet, consectetur adipiscing elit, sed do eiusmod tempor incididunt ut labore et dolore magna aliqua. Ut enim ad minim veniam, quis nostrud exercitation ullamco laboris nisi ut aliquip ex ea commodo consequat. Duis aute irure dolor in reprehenderit in voluptate velit esse cillum dolore eu fugiat nulla pariatur. Excepteur sint occaecat cupidatat non proident, sunt in culpa qui officia deserunt mollit anim id est laborum.")
```

SirBramble, 2023-02-21:
very useful! thanks

dejavu1987, 2023-04-18:
For me it is only 8 characters. Just wondering what is causing this limit. I have a similar workaround too but would be great to understand the actual cause and a fix for this.