T-vK/ESP32-BLE-Keyboard #172: Support mouse and keyboard profile

DevDavisNunez, 2022-06-26:
Could you make a profile that acts as a keyboard and mouse combo?

ywz978020607, 2022-07-13:
If a esp32 could act as mouse and keyboard meantime, it would be more useful!

WaGi-Coding, 2022-12-28:
I also look for a solution with a single BLE service that supports both the mouse and keyboard HID report descriptors, and use the same service to send and receive data for both devices. In this case, you would not need to disconnect and reconnect to switch between the mouse and keyboard.

I hope the creator can do this, i won't dive into BLE communication myself too deep ._. (aka. I am too stupid ^^)

alexz006, 2023-02-01:
> //Edit3: So i finally found a fork which supports all, even gamepad: https://github.com/juanmcasillas/ESP32-BLE-Combo

Does not work on iPhone 13 Pro Max.
And to make it work on Android, you need to change the line in the file BleCombo.cpp:
```
//pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND);
pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);
```

alexz006, 2023-02-01:
My fork with fixes and adding support for reading Numlock/Capslock/Scrolllock state
https://github.com/alexz006/ESP32-BLE-Combo