T-vK/ESP32-BLE-Keyboard #181: Change BLE Address

Frost159, 2022-08-22:
What I want to do is to be able, with the same esp32, to be able to choose at the start of the program to which device it should connect to (for example a windows pc and an android phone. 

Firstly I tried having 2 ble objects and choosing the object at the boot of the esp32, but even with different objects, my android device was still detecting the same BLE device and anto-connecting to it.
I also tried with different names but my android device was still autoconnecting whatever the name.
Is there a way to change a parameter so that it could be detected as 2 distincs devices.
I experimented a bit with BLE in the past and this parameter was the address, which I didn't found in the library.

anandameyer, 2022-12-13:
Currently I implemented the same mechanism to allow me to connect to different devices (auto reconnect) without turning off the other device's bluetooth.

Yes you can change mac address of bluetooth by calling `esp_base_mac_addr_set` before `.begin()` method. 

You can see the reference in [here](https://docs.espressif.com/projects/esp-idf/en/release-v3.0/api-reference/system/base_mac_address.html).

anandameyer, 2022-12-13:
> Is there a way to change a parameter so that it could be detected as 2 distincs devices.

Currently the device cannot appeared as 2 different devices, only 1 devices at a time with different mac address.