T-vK/ESP32-BLE-Keyboard #182: Doesn't send some special characters and the last charater go to loop

Alberto-00, 2022-08-29:
Hi, this library doesn't send "_" or other special characters like "!" or "$". Then i don't know when send text my esp32 print in loop the last character of the text that i send (actually i use the version 0.3.2)

ekoslav, 2022-08-29:
you can use ASCII codes to send keys:

    bleKeyboard.write(32);

If the last character goes into loop, either the speed of sending is too high, or you are sending too many chars at the same time.
The solution for latter is to send one char at the time using a loop.

Alberto-00, 2022-08-29:
> you can use ASCII codes to send keys:
> 
> ```
> bleKeyboard.write(32);
> ```
> 
> If the last character goes into loop, either the speed of sending is too high, or you are sending too many chars at the same time. The solution for latter is to send one char at the time using a loop.

I solved the loop problem but the special character are not shown with ASCII mode. I tried in this way to show "_": 
```
bleKeyboard.write(95);
bleKeyboard.write(0x5F);
```

ekoslav, 2022-08-29:
Not sure why it happens to you, I just tried quick on my password keyboard ( I am at work now, so no time for testing).
It works wit "_" char with no issues whatsoever.
You can take a look ant the code for it: https://github.com/ekoslav/ESP32_Password_keyboard/blob/main/ESP32_4x4MatKBD_SPIFS/ESP32_4x4MatKBD_SPIFS.ino

it is under "**void SendSequence(String sequence)**" - This receives a string and sends it to a BT. It has some special sequences like \r or \s for CR and SPACE

Alberto-00, 2022-08-29:
> Non sono sicuro del motivo per cui ti succede, ho appena provato rapidamente sulla tastiera della mia password (sono al lavoro ora, quindi non ho tempo per i test). Funziona con il carattere "_" senza alcun problema. Puoi dare un'occhiata al codice: [https://github.com/ekoslav/ESP32_Password_keyboard/blob/main/ESP32_4x4MatKBD_SPIFS/ESP32_4x4MatKBD_SPIFS.ino ](https://github.com/ekoslav/ESP32_Password_keyboard/blob/main/ESP32_4x4MatKBD_SPIFS/ESP32_4x4MatKBD_SPIFS.ino)
> 
> è in " **void SendSequence(String sequence) **" - Questo riceve una stringa e la invia a un BT. Ha alcune sequenze speciali come \r o \s per CR e SPACE

Now I try to set delay of 200 ms

Alberto-00, 2022-08-29:
> When I have issue with speed, it happens to replace "_" char with "-". I am not sure if this is relevant.

The strange thing is that some ASCII code in decimal aren't decoded in the right code. For example: "(" --> 40 in decimal but my device translate its in ")"

ekoslav, 2022-08-29:
Even more wierd is that this happens (for me) only in RDP sessions, especially if it goes into lock screen. Also in this case copy/paste also does not work, and the Bluetooth keyboard does work if I slow it down to looks like it is being manually typed.
This points to be a special case, not strictly an issue with the library.

Alberto-00, 2022-08-29:
> Even more wierd is that this happens (for me) only in RDP sessions, especially if it goes into lock screen. Also in this case copy/paste also does not work, and the Bluetooth keyboard does work if I [...]

So how can i solve this problem?

ekoslav, 2022-08-29:
Maybe try to copy the above mentioned SendSequence , and see if it will work for you. I am using it daily and have no issues with it on several different machines.

If not, it may have some compatibility issue with the device the ESP32 BT is attached to .

Alberto-00, 2022-08-30:
> Maybe try to copy the above mentioned SendSequence , and see if it will work for you. I am using it daily and have no issues with it on several different machines.
> 
> If not, it may have some compatibility issue with the device the ESP32 BT is attached to .

I used your function. In serial console the special characters are printed but the bluetooth print or write functions don't print some special characters but print "?".

Alberto-00, 2022-08-30:
Why these characters match to this ascii code:
33 !
34 °
35 £
36 $
37 %
38 /
39 à
40 )
41 =
42 (
43 ^
44 ,
45 '
46 .
47 -
58 ç
59 ò
60 ;
61 ì
62 :
63 _
64 "
91 è
92 ù
93 +
94 &
95 ?
96 \
123 é
124 §
125 *
126 |
190 ,
236 <

ekoslav, 2022-08-30:
What is the device your ESP32 is connected to? Have you tried it on any other device?

Alberto-00, 2022-08-30:
> What is the device your ESP32 is connected to? Have you tried it on any other device?

The problem is that the library functions correctly on English keyboard

samagas, 2022-12-07:
As far as I know, the original arduino keyboard library uses "coordinates" to send the key strokes, so when you send a character what the code actually does is to translate the character ASCII code to the actual key you would need to press to generate it in the US layout. It is really likely that you may see some characters/symbols on your keyboard but they are different on the US keyboard.

The arduino version lets you pick the language of the keyboard so you can have the appropiate mapping, I've not seen this mentioned anywhere on this code but I'm pretty sure you can mimic it.

Open the "BleKeyboard.cpp" file and around line 220 you'll see:
`const uint8_t _asciimap[128] =
{
	0x00,             // NUL`

You can find the corresponding mapping for It and change the values there.