T-vK/ESP32-BLE-Keyboard #186: Power off reconnection

1998-zhang, 2022-09-02:
How to save connection information before power failure。Prevent reconnection every time。

anandameyer, 2022-12-13:
What you mean with reconnection? Do you mean to avoid another device to do reconnection to ESP  or avoid ESP to do reconnection to another device? 

As long as I know, ESP will save all bonded device in its nvs partition, and this library uses BLE Server which is not doing reconnection but advertisement. 

Short, you never need to save connection.

mayaku2go, 2023-01-21:
I had the same problem and created just an account for this.
Here is the solution:
https://github.com/nkolban/esp32-snippets/issues/632

Open bleKeyboard.cpp and add the two lines  to the begin function.
```
uint8_t val[] = {0x01, 0x00};
desc->setValue(val, 2);
```

The whole function should look like this:
```
void BleKeyboard::begin(void)
{
  BLEDevice::init(deviceName);
  BLEServer* pServer = BLEDevice::createServer();
  pServer->setCallbacks(this);

  hid = new BLEHIDDevice(pServer);
  inputKeyboard = hid->inputReport(KEYBOARD_ID);  // <-- input REPORTID from report map
  outputKeyboard = hid->outputReport(KEYBOARD_ID);
  inputMediaKeys = hid->inputReport(MEDIA_KEYS_ID);

  outputKeyboard->setCallbacks(this);

  hid->manufacturer()->setValue(deviceManufacturer);

  hid->pnp(0x02, vid, pid, version);
  hid->hidInfo(0x00, 0x01);

#if defined(USE_NIMBLE)

  BLEDevice::setSecurityAuth(true, true, true);

#else

  BLESecurity* pSecurity = new BLESecurity();
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND);

#endif // USE_NIMBLE

  hid->reportMap((uint8_t*)_hidReportDescriptor, sizeof(_hidReportDescriptor));
  hid->startServices();

  onStarted(pServer);

  advertising = pServer->getAdvertising();
  advertising->setAppearance(HID_KEYBOARD);
  advertising->addServiceUUID(hid->hidService()->getUUID());
  advertising->setScanResponse(false);
  advertising->start();
  hid->setBatteryLevel(batteryLevel);

  ESP_LOGD(LOG_TAG, "Advertising started!");
}
```