T-vK/ESP32-BLE-Keyboard #187: LIbray Size

saulc, 2022-09-03:
I used this to convert one of my project from usb hid to ble. its working great but the big problem is the compilation size is 
over 80% (~1066721 bytes) on a feather v2 esp32. 

Which means I can't even include the sleep or even wifi lib to disable it and reduce power use. 
current battery life on 600ma is bad.  

Anyway to do a more lightweight version?

ekoslav, 2022-09-03:
Have you tried different partition scheme?

[image]

saulc, 2022-09-03:
Thanks! it seemed weird to hit the memory limit on this thing...I only had 2 options for my board 
[image]

1075517 bytes (51%) - > 1454461 bytes (69%) with wifi disabled! 

Nice work on the library. switching from hid keyboard was just find/replace and then add a device name and update the battery %.

Now that I think about it is there any way to show charging or low battery?

T-vK, 2022-09-04:
Check the Readme for NimBLE-Mode.