T-vK/ESP32-BLE-Keyboard #188: ESP-WROOM-32 Bluetooth connection flickering during pairing

cnoronh, 2022-09-03:
When running the example program (SendKeyStrokes), the ESP-WROOM-32 Bluetooth keeps flickering (device appears briefly and disappears when paring with Windows 10).  I am unable to pair unless I time it perfectly.  Does anyone know what could cause this/how to fix this?  I do not see this issue when running this sketch with the ESP-WROOM-32D.

ekoslav, 2022-09-04:
I've seen this issue on some W10 / W11 machines. Not sure what it causes it, but if you leave it long enough, it will connect. The quick solution is to remove device, turn BT off, wait 15 -20 seconds, turn it on, and connect device.
I noticed it happens after ESP32 is powered off without disconnecting bt connection.

I don't know if helps, I het a lot of these messages while flip-flopping:
"lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)"

JamesCohen-awin, 2023-01-19:
Out of interest was your sketch compiled with or without NimBLE?

cnoronh, 2023-01-20:
I did not compile with NimBLE.  This issue does not arise with the ESP-WROOM-32D so my recommendation is to use this model at minimum when using this library to avoid any headache.

pranavpa8788, 2023-06-20:
I had the same issue with ESP32 C3 DEV KIT M1. I solved this by connecting once to my android device -> turning off my android's bluetooth and reconnecting on my PC. Don't know why this works. But before this I was able to connect, but the keystrokes wouldn't work, now it works properly (I'm on windows 11 and my android version is 12)

babelmandeb124, 2024-02-03:
same issue esp32 devKit v3, win 11