T-vK/ESP32-BLE-Keyboard #192: Can't connect to Android

vcchm, 2022-10-01:
-I am using a Samsung Note 20 Android Phone.
-The Esp device is a Lily TKeyboard with an ESP32C3 chip
-The following code can't connect to my Android phone
-On the android I have a message "Impossible to connect a new device"
-Code used is the follwing simple code - it works perfectly with my PC

/**
 * This example turns the ESP32 into a Bluetooth LE keyboard that writes the words, presses Enter, presses a media key and then Ctrl+Alt+Delete
 */
#include <BleKeyboard.h>

BleKeyboard bleKeyboard;

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleKeyboard.begin();
}

void loop() {
  if(bleKeyboard.isConnected()) {
    Serial.println("Sending 'Hello world'...");
    bleKeyboard.print("Hello world");

    delay(1000);

    Serial.println("Sending Enter key...");
    bleKeyboard.write(KEY_RETURN);

    delay(1000);

    Serial.println("Sending Play/Pause media key...");
    bleKeyboard.write(KEY_MEDIA_PLAY_PAUSE);

    delay(1000);

   //
   // Below is an example of pressing multiple keyboard modifiers 
   // which by default is commented out.
    /*
    Serial.println("Sending Ctrl+Alt+Delete...");
    bleKeyboard.press(KEY_LEFT_CTRL);
    bleKeyboard.press(KEY_LEFT_ALT);
    bleKeyboard.press(KEY_DELETE);
    delay(100);
    bleKeyboard.releaseAll();
    */
  }

  Serial.println("Waiting 5 seconds...");
  delay(5000);
}

vcchm, 2022-10-06:
When I use the library 0.3.0 it pairs & works the first time. But when I switch off and on the ESP32, it pairs but the keyboard is not working anymore...

joostbijl, 2022-11-07:
The fix with regard to setting a different authentication method (#166 ) works for most scenarios as far as i can see. I suggest to change that in your local copy of the ble keyboard library

This is the change in My Documents\Arduino\libraries\ESP32-BLE-Keyboard-0.3.2-beta\BleKeyboard.cpp. On line 130 change pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND); to pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);

I'm not really sure what these authentication modes are and how they affect other systems. But so far so good.

vcchm, 2022-11-11:
Thanks. I am using now the last stable version, and generating a new name & Mac adress for each compilation. Then it works for win 10, win 11 and android as well as IPhone.

panakos, 2023-01-24:
> The fix with regard to setting a different authentication method (#166 ) works for most scenarios as far as i can see. I suggest to change that in your local copy of the ble keyboard library
>
> This [...]

This works on all my android devices except lenovo tab m7 that's running android go .it connects 1st time but will not reconnect any asks to be paired again

mayaku2go, 2023-01-25:
This should help:
https://github.com/mayaku2go/ESP32-BLE-Keyboard/releases/tag/v0.3.2b

panakos, 2023-01-25:
> This should help: https://github.com/mayaku2go/ESP32-BLE-Keyboard/releases/tag/v0.3.2b

Thanks , I'll give it a try!

panakos, 2023-01-31:
> > This should help: https://github.com/mayaku2go/ESP32-BLE-Keyboard/releases/tag/v0.3.2b
> 
> Thanks , I'll give it a try!

Unfortunately it's doing the same . It will work on pairing and will reconnect if you turn off the tablets bluetooth or if you reset the esp32 , but it will ask to be paired again if you power cycle the esp32

gator55, 2024-05-22:
> The fix with regard to setting a different authentication method (#166 ) works for most scenarios as far as i can see. I suggest to change that in your local copy of the ble keyboard library
>
> This [...]

Now volume control/play/pause works with Xiaomi. Many thanks!