T-vK/ESP32-BLE-Keyboard #194: Can connect OK, but nothing happens (iOS and MacOS)

troglodisme, 2022-10-06:
Just installed the latest version of the library and testing the example sketch to try and write something to an iOS device or MacOS M1 but it doesn't seem to do anything. 

I can see the devices and I can pair to them, but then no text (e.g. "Hello world") is typed. I have tried other commands (volume etc.. ) but still nothing happens. 

I am using an ESP32-S3 Mini and iPhone 12 mini running iOS16 and MacBook Air M1 running Monterey (12.5). 

Has anyone faced the same issue?

Corgano, 2022-10-18:
I'm getting the same thing on windows 11. Same module, same output to console, but nothing typing

Mew463, 2022-11-03:
#169 Worked for me.

Seref, 2022-11-04:
> #169 Worked for me.

It still disconnects for me on PC and only works some times with this fix and often causes a crash :(

stephan2307, 2023-02-21:
I too have this issue. ESP DEVKIT V1 board. Connects fine to Android 12 ( Fairphone 4 ) but nothing is happening on the phone. Console shows the script is running though.

Any ideas?