T-vK/ESP32-BLE-Keyboard #207: How to add an Android specific action keys

llilakoblock, 2023-01-26:
Hi and thanks for a great library! It works flawlessly with my project. 
But I still cant understand logic behind key encoding. So basically, I want to add specific Android keystroke to be able to send it. Like this one https://developer.android.com/reference/android/view/KeyEvent#KEYCODE_VOICE_ASSIST

I have read suggested answers in another issue, but they are IOS specific. Any help would be appreciated!

mayaku2go, 2023-01-26:
Maybe this is helpful:
[https://www.usb.org/sites/default/files/voice_command_usage.pdf](url)
But only adding this value to the hidReportDescriptor isn't enough, I guess.
I got the info above from here:
[https://usb.org/sites/default/files/hut1_3_0.pdf](url)

mayaku2go, 2023-01-26:
Here is a complete tutorial about custom HID report descriptors:
[https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/](url)

llilakoblock, 2023-01-26:
> Here is a complete tutorial about custom HID report descriptors: [https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/](url)

Thanks! Will try to dive in it 👍

llilakoblock, 2023-01-27:
So, for anyone wondering how to make it done.

For key event KEYCODE_VOICE_ASSIST to be understood right by Android, we must implement Headphone HID device, it in Consumer section with four functions, according to Android ref and USB docs: https://source.android.com/docs/core/interaction/accessories/headset/usb-headset-spec
https://usb.org/sites/default/files/hut1_2.pdf

Extending HID report in BleKeyboard.cpp:

[image]

Extending MediaKeyReport in BleKeyboard.h:

[image]

And registering our new Report ID in begin() func:

[image]

Issue may be closed.

roberto1963, 2023-02-24:
Hello, 
This library is really great, it works well for texting and taking a photo (with KEY MEDIA VOLUME_UP)
It would be great to add the keycodes to open the Android camera app and other possibilities as shown in this link : 
https://elementalx.org/button-mapper/android-key-codes/
I have read the recommended documents and I tried to understand how HID descriptors work but I admit that it's too complicated for me... 8-((
Is there anyone who could help me write the missing code ? 
I would be happy to do all the tests.