T-vK/ESP32-BLE-Keyboard #210: setBatteryLevel() and USE_NIMBLE

alexz006, 2023-02-02:
If you set the battery level to setBatteryLevel() after bigin(), and if NIMBLE is not used, then everything works as it should and the battery level is updated in Windows 11.

If you use NIMBLE, the charge level is not updated in Windows 11.

[image]

```C++
/*
  Tests setBatteryLevel() if

  #define USE_NIMBLE
  and
  // #define USE_NIMBLE
  
  in BleKeyboard.h
*/

#include <BleKeyboard.h>

BleKeyboard bleKeyboard;

uint8_t BatteryLevel = 100;
unsigned long previousMillisBattery = 0; 

void setup() {
  Serial.begin(115200);

  bleKeyboard.setBatteryLevel(BatteryLevel);
  Serial.println("Battery Level before begin(): " + String(BatteryLevel));
  
  bleKeyboard.begin();

  while (!bleKeyboard.isConnected()) {
    Serial.print(".");
    delay(250);
  } Serial.println();

  delay(5000);

  BatteryLevel--;
  bleKeyboard.setBatteryLevel(BatteryLevel);
  Serial.println("Battery Level after begin(): " + String(BatteryLevel));
}

void loop() {
  while (!bleKeyboard.isConnected()) {
    Serial.print(".");
    delay(250);
  }

  // TEST setBatteryLevel()
  unsigned long currentMillis = millis();
  if(BatteryLevel>=4 && currentMillis - previousMillisBattery >= 3000) { // gradual discharge of the battery
    previousMillisBattery = currentMillis;
    BatteryLevel--;
    bleKeyboard.setBatteryLevel(BatteryLevel); // works only #if !defined(USE_NIMBLE)
    bleKeyboard.begin();
    Serial.println("BatteryLevel in loop(): " + String(BatteryLevel));
  }

}
```

alexz006, 2023-02-22:
https://github.com/h2zero/NimBLE-Arduino/pull/422

battosai30, 2023-09-30:
Same problem here :)

jojo-06, 2024-03-13:
adding `m_batteryLevelCharacteristic->notify();` to the setBatteryLevel function worked for me:
(in file NimBLEHIDDevice.cpp)
````
void NimBLEHIDDevice::setBatteryLevel(uint8_t level) {
	m_batteryLevelCharacteristic->setValue(&level, 1);  
	m_batteryLevelCharacteristic->notify(); // new  
}
````
hope this helps

ignis32, 2024-05-12:
@jojo-06 

Thanks a lot, that was useful and helped a lot