T-vK/ESP32-BLE-Keyboard #212: after arduino shutsdown must re-pair to work properly...

lf1969, 2023-02-04:
hi.

if shutdown the arduino while a connection is established, when arduino is restarted, the arduino says is connected and sending 
messages, the host device is also connected but messages are not processed.

if I upair/pair, it works again. after the pairing if I restart the BT connection for the host, all is fine.

Just cant stop the arduino :(
.

ekoslav, 2023-02-04:
Try using NimBLE mode, it worked for me  with the same problem.

lf1969, 2023-02-06:
Hi.

Thanks for your reply.

You referring to:

https://github.com/h2zero/NimBLE-Arduino

ekoslav, 2023-02-06:
This library has it.

[https://github.com/T-vK/ESP32-BLE-Keyboard#nimble-mode](https://github.com/T-vK/ESP32-BLE-Keyboard:)

> NimBLE-Mode
> The NimBLE mode enables a significant saving of RAM and FLASH memory.

[Hot to activate NimBLE mode](https://github.com/T-vK/ESP32-BLE-Keyboard#how-to-activate-nimble-mode)

lf1969, 2023-02-06:
Hi.

Sorry to bother you again.

I've enabled nimble, flashed my chip with latest firmware, still not
working properly.

I'm using this board:

Teyleten Robot ESP32S ESP32 ESP-WROOM-32 Development Board 2.4GHz Dual-Core
WiFi +Bluetooth 2 Function Microcontroller for Arduino (ESP32 38P, 3PCS)
https://a.co/d/hkgr0ai

If you like I can send you money to purchase a board to determine why this
is an issue?

ekoslav, 2023-02-07:
I am not sure if you noticed, to enable NimBLE you need to make a change in the library file, not your application. 
To test it yourself, I think it is probably easiest is to temporary flash one of your boards with the tested to work software ( without compiling).
Here is the link to my project that is tested to automatically re-connect to Win 11:  [https://www.tenpurplebirds.com/?page_id=375](https://www.tenpurplebirds.com/?page_id=375)

[image]
Select the version 1.13, and flash it over the web. This way, if it does work, there is an issue with you library setup. Otherwise it may be the compatibility of your board.
This way test cost nothing but about 10 minutes of your time :)

ekoslav, 2023-02-07:
I forgot to mention that you do not need to attach anything to ESP32 board. Once booted it should show up as this:
[image]
The last 4 digits in the name a re specific to a board MAC. 
Once paired, you can power cycle ESP or the Windows machine, and see if they connect automatically. On the ESP you can monitor events in the serial console.

lf1969, 2023-02-07:
Thanks for the tip!

joostbijl, 2023-02-14:
I had the same issues, see #166 for a way that works for me. 

I never became friends with NimBLE so far

lf1969, 2023-02-23:
Thanks for getting back to me.

Its related to having ble classes in multiple locations and to make matters
worse if I use a different board it works.

luissperber, 2023-08-10:
> I am not sure if you noticed, to enable NimBLE you need to make a change in the library file, not your application. To test it yourself, I think it is probably easiest is to temporary flash one of [...]

> This library has it.
>
> [https://github.com/T-vK/ESP32-BLE-Keyboard#nimble-mode](https://github.com/T-vK/ESP32-BLE-Keyboard:)
>
>
> [Hot to activate NimBLE [...]

Works perfectly! Could you tell me what version of NinBLE is used and what version of BLE-Keyboard? Is there another secret?

pookzilla, 2024-04-14:
For those who are having issues with this I found switching to NimBLE and replacing `BLEDevice::setSecurityAuth(true, true, true);` with `BLEDevice::setSecurityAuth(true, false, false);` in BleKeyboard.cpp fixed my issues.  I can now power cycle the device and it will reconnect.