T-vK/ESP32-BLE-Keyboard #213: Mic mute

s56jsc, 2023-02-04:
Great library!
I only miss microphone mute key? Can be added also this?

ekoslav, 2023-02-04:
On Win 11 you can just send win+alt+k

s56jsc, 2023-02-04:
I'm talking for androide phone.

ekoslav, 2023-02-04:
That is an interesting idea. This library is the BT version of the standard keyboard. A quick google search did not return useful solution. Due to my work requirements, I always wanted a BT headset with the mute button, but I found only one, and that particular one did not send the "mute" command to the phone, but muted microphone internally. 
So this one might be a bit tough.
Now, there are some BT selfie buttons, to trigger camera they act as a keyboard, and send letter "S" or "K" (not sure about exact letters, it has been a while since I last tried it, and don't have one around now to test).
So maybe plug in the full size USB keyboard to your android device and see if any key, or key combination triggers "mute" action. You can also test sending "ALT" codes directly by holding key "ALT" and 3 digit code on the numeric keyboard, and the releasing ALT key ( example: (ALT)064(release ALT) will produce sign @).
If you do succeed, please post results here, I curious to know if it actually  exists.

s56jsc, 2023-02-05:
I have one BT keyboard got tablet. But or, keyboard or android system not support ALT+ digit. There nothing happens.
I also try combination, but not any luck for now. I only found, that if you are in call and double press SPACE it will mute microphone.
This is some kind of solution. It work on my sony XZ2. I don't know for other phones. Please try it and report.