T-vK/ESP32-BLE-Keyboard #225: can't make it work with Android TVBox

simon-schvartzman, 2023-03-08:
Hi team, I'd really like this solution to use an ESP32 as RC for my TvBox (Android 10.0) 
Congratulations and many thanks to the developers.

Now the problem: I was able to pair the ESP with the TVBox and it is listed under "Remotes & Accessories".

Using the SendKeyStrokes example nothing gets keyed in while using a very simple text edit (Notes)
I can see that SendKeyStrokes is executing the loop because Sending 'Hello World'... is shown in the monitor meaning **bleKeyboard.isConnected()** as far as I understand

What could be wrong? Am I missing something?

Any help will be greatly appreciated, many thanks in advance

simon-schvartzman, 2023-03-10:
After reseting both the TVBox and the ESP it is working now...

jakaisna, 2023-06-27:
Try this fork: https://github.com/alexz006/ESP32-BLE-Combo