T-vK/ESP32-BLE-Keyboard #233: bluetooth is  undetected on my windows 10 laptop (BT version 3.x)

r4ph74, 2023-05-11:
hi , I have downloada install the latest beta driver 0.3.2 and 0.3.3 . and alos tried stable verison 0.30 

all of them doesn't work , My Windows 10 , unable to detect it .  But it is detected on my Android Phone (redmi note 8 Pro). 

I'm using esp32- Nodemcu  - Ble version 

what should i do so we enable to detect the esp32S Nodemcu ?

alexz006, 2023-05-20:
Try this fork: https://github.com/alexz006/ESP32-BLE-Combo

r4ph74, 2023-05-21:
Thanks... Alex , I will try .. now .

r4ph74, 2023-05-21:
D:\Scorpion\Project\3Dprinting\T3D\Mpad\Sketch\ble_keypad_combo\ble_keypad_combo.ino: In function 'void setup()':
ble_keypad_combo:11:13: error: 'class BleCombo' has no member named 'setManufacturer'
   bleDevice.setManufacturer("Espressif"); //call before any of the begin functions to change the Manufacturer name.
             ^
exit status 1
'class BleCombo' has no member named 'setManufacturer'

alexz006, 2023-05-21:
You need to copy the library [ESP32-BLE-Combo](https://github.com/alexz006/ESP32-BLE-Combo) to the `libraries` folder, and move the ESP32-BLE-Keyboard library outside of the `libraries` folder to avoid a conflict of libraries.

r4ph74, 2023-05-21:
[image]

I only have these in my library folder. No more Esp32-BLE-Keyboard anymore.

r4ph74, 2023-05-21:
I have tried upload and test the basic send keystrok example provided.  

my android phone able to detects and pair the esp32 , buat when  I open Google keep note, it shows nothing, even in the serial monitor .. sending text and running the script normally,

on the other hand , my windows 10 still unable to detect my esp32.

alexz006, 2023-05-21:
> [image]
> 
> I only have these in my library folder. No more Esp32-BLE-Keyboard anymore.

You need to move the `Keypad` folder somewhere else, so that only `ESP32-BLE-Combo` remains in the libraries folder without `Keypad`!

r4ph74, 2023-05-21:
Oh I see, I will try again Thanks