T-vK/ESP32-BLE-Keyboard #234: russian symbols

kraa965, 2023-05-17:
hi, tell me, is it possible to implement the output of Russian characters and, if possible, explain how this can be done?

alexz006, 2023-05-20:
ESP sends a "physical" key press, just like on a regular keyboard. Therefore, to input Russian characters (or any other language), you need to send a key combination to switch the keyboard layout.

kraa965, 2023-05-21:
This is how Russian characters are printed if you change the language
[image]
But if you write the code in this way, then everything is output correctly
[image]
[image]
but there is a task so that the user can use both languages, so that it is possible to implement this without switching languages?

alexz006, 2023-05-21:
You cannot write Russian text using blekeyboard.println(). As an alternative, translate Cyrillic characters into Latin characters, as on a keyboard, and then perform blekeyboard.println().
Something like this:
```c++
void Keyboard_println(String text) {
  delay(200);
  String latinText = "";
  for (int i = 0; i < text.length(); i++) {
    char cyrillicChar = text.charAt(i);
    char latinChar;
    
    switch (cyrillicChar) {
      case 'а':
        latinChar = 'f';
        break;
      case 'б':
        latinChar = ',';
        break;
      // add the rest of the symbols and their correspondences here
      default:
        // If the symbol is not found, leave it unchanged.
        latinChar = cyrillicChar;
        break;
    }
    
    latinText += latinChar;
  }
  
  blekeyboard.println(latinText);
}
```

kraa965, 2023-05-21:
> You cannot write Russian text using blekeyboard.println(). As an alternative, translate Cyrillic characters into Latin characters, as on a keyboard, and then perform blekeyboard.println(). Something [...]

thank you, and if the text is passed not as a String, but as a char *text?