T-vK/ESP32-BLE-Keyboard #235: Sending 'back key' doesn't seem to work

joostbijl, 2023-05-25:
Hi,

i want to send the 'back key' as found on Android devices via BT HID. According to https://developer.android.com/reference/android/view/KeyEvent#KEYCODE_BACK the keycode is 0x04.

So i defined a constant like this in my code: `const uint8_t KEY_BACK = 0x04;`. 

However, if i `write()` this via BT it simply doesn't arrive according to KeyEventDisplay (and the apps don't go 'back').

I'm probably missing something here ;) - How can i send the 'back' key?

thx

lubaldi, 2023-11-21:
do you find a solution?

Guillemete, 2024-03-19:
You have to replace one of those command with your "back" command in the BleKeyboard.cpp file and the .h file :
(in this exemple, if you want to replace the STOP command with BACK)

so, in the BleKeyboard.cpp, look for this

USAGE(1),           0xB7,          //   USAGE (Stop)                ; bit 2: 4
USAGE(1),           0xCD,          //   USAGE (Play/Pause)          ; bit 3: 8

AND in the BleKeyboard.h

const MediaKeyReport KEY_MEDIA_STOP = {4, 0};
const MediaKeyReport KEY_MEDIA_PLAY_PAUSE = {8, 0};

------------------------

and replace with : 

USAGE(1),           0x04,          //   USAGE (back)                ; bit 2: 4

const MediaKeyReport KEY_MEDIA_BACK = {4, 0};

it should work, assumin that you have the right ble code