T-vK/ESP32-BLE-Keyboard #245: A fatal error occurred: Serial data stream stopped: Possible serial noise or corruption

LLLCDYY, 2023-07-29:
Sometimes this situation will suddenly occurs
`Processing nodemcu-32s (platform: espressif32; board: nodemcu-32s; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 (5.1.0) > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20004.0 (2.0.4)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32 BLE Keyboard @ 0.3.2
Building in release mode
Retrieving maximum program size .pio\build\nodemcu-32s\firmware.elf
Checking size .pio\build\nodemcu-32s\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  10.0% (used 32756 bytes from 327680 bytes)
Flash: [========  ]  80.6% (used 1055789 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM5
Uploading .pio\build\nodemcu-32s\firmware.bin
esptool.py v3.3
Serial port COM5
Connecting.....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:d7:eb:0b:25:44
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00113fff...
Flash params set to 0x0220
Compressed 17408 bytes to 12093...
Writing at 0x00001000... (100 %)
Wrote 17408 bytes (12093 compressed) at 0x00001000 in 0.7 seconds (effective 212.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 276.1 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.2 seconds (effective 414.8 kbit/s)...
Hash of data verified.
Compressed 1061568 bytes to 669140...
Writing at 0x00010000... (2 %)
Writing at 0x0001b2a8... (4 %)
Writing at 0x00026e94... (7 %)
Writing at 0x00036514... (9 %)
Writing at 0x0003ba99... (12 %)
Writing at 0x00041209... (14 %)
Writing at 0x000470a8... (17 %)
Writing at 0x0004caed... (19 %)
Writing at 0x00051d56... (21 %)
Writing at 0x00057519... (24 %)
Writing at 0x0005d34a... (26 %)
Writing at 0x00062954... (29 %)
Writing at 0x00067dab... (31 %)
Writing at 0x0006d29e... (34 %)
Writing at 0x00073220... (36 %)
Writing at 0x00078939... (39 %)
Writing at 0x0007e2fd... (41 %)
Writing at 0x00084306... (43 %)
Writing at 0x00089dc5... (46 %)
Writing at 0x0008f39a... (48 %)
Writing at 0x00094e9c... (51 %)
Writing at 0x0009a4a1... (53 %)
Writing at 0x0009fc7d... (56 %)
Writing at 0x000a5a0c... (58 %)
Writing at 0x000abc36... (60 %)
Writing at 0x000b2b9d... (63 %)
Writing at 0x000b8516... (65 %)

A fatal error occurred: Serial data stream stopped: Possible serial noise or corruption.
*** [upload] Error 2`

**please help me,There are no issues with the data cable or the core board.**

LLLCDYY, 2023-07-29:
And I have already tried replacing the data cable

animevietsub, 2023-07-29:
Please ensure you don't connect these pins while programming GPIO0, GPIO5, GPIO12, GPIO14, and GPIO15.
If you don't connect anything to those pins check RX, TX signal.

LLLCDYY, 2023-07-30:
> 请确保在编程 GPIO0、GPIO2、GPIO5、GPIO12、GPIO14 和 GPIO15 时不要连接这些引脚。如果您没有将任何东西连接到这些引脚，请检查 RX、TX 信号。如果需要连接到这些引脚，请检查以下条件： GPIO 0（必须为低才能进入启动模式） GPIO 2（在启动期间必须为浮动或低） GPIO 5（在启动期间必须为高） GPIO 12（在启动期间必须为**低） GPIO 15（在启动期间必须为高）**

I should have discovered the problem, it should be a problem with the flash memory chip. If the code size is relatively large, it will have problems. I have this issue with both of my core boards, so I replaced them with a brand new core board and there was no problem.